IFL-CAMP/iiwa_stack #1: collaboration with grl

ahundt, 2016-02-16:
Hey,

I'm also working on an iiwa stack though I'm taking a slightly different approach. I was wondering if you might be interested in some degree of collaboration and/or enabling some integration between our setups?

My project is grl:
https://github.com/ahundt/grl

The bulk of the kuka code can be found at:
https://github.com/ahundt/grl/tree/master/include/grl/kuka

We are working towards a [driver supporting FRI and Java](https://github.com/ahundt/grl/blob/master/include/grl/kuka/KukaDriver.hpp#L38) at the user's option. My group and I also have the first implementation of a [ROS driver for this purpose](https://github.com/ahundt/grl/blob/master/include/grl/ros/KukaLBRiiwaROSPlugin.hpp). 

I took a slightly different approach from yours so that users don't need ROS to use the driver. My system uses ZeroMQ and google flatbuffers to send messages to the JAVA API. When in FRI mode it directly sends packets to the FRI system but is planned to also be able to simultaneously communicate with the JAVA side for ongoing configuration changes. 

Since using the KONI interface for the JAVA side is exclusive to using the FRI I currently don't take that step. What is the advantage there?

I'm also interested in your integration with other toolsets like moveit and gazebo.

Thanks for taking the time to consider my thoughts!

SalvoVirga, 2016-02-17:
Hi Andrew,

thanks a lot for the interest, we heard already about your nice work from our colleagues at JHU.

I come right away at the point about the KONI port : by design that port is meant to be used only for FRI, correct. Without any modification that port is handled directly from the Real Time OS on board of the KUKA controller, the Java calls for FRI (with that .setOverlay on a motion) do the magic so that commands send though FRI are executed. But that also means, to my knowledge, than you cannot send other than FRI messages using that port, as we don't have direct access to the RTOS.

The modification that we describe in our wiki, basically assigns the KONI port to the Windows7 Embedded on the cabinet, so that it is accessible from Java. This change comes from the original work of Stanford : https://bitbucket.org/khansari/iiwa that we initially followed. But in theory is not necessary, as one could simply use the same ethernet port used for Sunrise Workbench. Also, FRI allows commands at 1ms because it is linked to the RTOS, with that change we lose that too.

At the moment, we use the KONI port for commanding the robot using KUKA's ServoMotion and for a dynamical configuration of the system (most of that is on our development branch). Our plan in the near future is to do all this using the Sunrise port, reenable the KONI port to be used with FRI and so allow a user to choose between different command modes.

As a matter of principle, of course we are open to any collaboration and any help or suggestion is welcome. But in the specific case, it is hard for us to find a submodule that could be neatly shared through a common interface between our two different approaches. We plan to continue using ROS topics and services, and in future actions, for communication because their expressiveness. Apart from that, the rest is actually just a thin wrapper around the Sunrise APIs, so providing an interface on top of it would be a huge overhead with little benefits.

On MoveIt! and Gazebo, if I remember correctly you use a V-REP model of the robot, while those two modules need an URDF model to work. If you plan to use MoveIt! it might make sense to switch to that representation, I also don't have knowledge of the accuracy of the V-REP model to be honest (our URDF model leads to a forward kinematics that coincides to that from Sunrise up to a micron).

Any idea about what we could share?

ahundt, 2016-02-24:
@SalvoVirga Based on our conversation via video conference I wanted to make some notes:

- You can at least update the license file for BSD (hopefully today if possible?) and just provide written comment here saying it is ok.
- grl can use the ros configuration, URDF, etc from this repository
- I'll aim to provide the same ros APIs (topics, etc) you are using in C++
- I'll try to integrate your functionality where appropriate
- We will discuss further integration between our tools after your March 1 deadline

grl main kuka driver code and locations:
- **Independent** (no V-REP) [Kuka C++ driver classes](https://github.com/ahundt/grl/tree/master/include/grl/kuka)
    - smart servo
    - FRI
    - configuration
- [ROS driver](https://github.com/ahundt/grl/blob/master/include/grl/ros/KukaLBRiiwaROSPlugin.hpp)
    - in progress
    - depends on kuka drivers
- [V-REP drivers](https://github.com/ahundt/grl/tree/master/include/grl/vrep)
    - Kuka V-REP class depends kuka drivers
    - Includes hand eye calibration from camodocal

All that sound good? Thanks for the presentation it was extremely informative!

ahundt, 2016-02-24:
The code there isn't quite the latest, I'm in the midst of integration and I push changes out as it makes sense.

SalvoVirga, 2016-02-24:
@ahundt Changed the main LICENSE file to BSD in the master branch, I will modify the single modules later on. You can consider the whole project under BSD from now on.

ahundt, 2016-03-03:
@SalvoVirga Now that some key deadlines are past I was hoping to follow up on this if you're available. Feel free to send an email to athundt@gmail.com and we can set something up.

marcoesposito1988, 2016-03-11:
Since such a topic is better suited to a private conversation, I am closing this issue.

closed 2016-03-11 by marcoesposito1988

ahundt, 2016-03-19:
@marcoesposito1988 @SalvoVirga I've released [grl 3.0.0](https://github.com/ahundt/grl/releases/tag/v3.0.0) which includes partial integration with iiwa_stack! If you are interested in adding FRI support to iiwa_stack I provide [classes designed specifically for use by other library authors](https://github.com/ahundt/grl/tree/v3.0.0/include/grl/kuka). I'm also using an approach for communication with the controller that avoids https://github.com/SalvoVirga/iiwa_stack/issues/7.

I also have a decent approach to supporting ROS