IFL-CAMP/iiwa_stack #3: complete list of dependencies?

ahundt, 2016-02-28:
There seems to be a lot of dependencies for this that aren't documented. Could I get a complete list?

So far I've needed to add:

- from [ros-controls](https://github.com/ros-controls)
    - ros_control
    - ros_controllers
    - realtime_tools
    - control_toolbox

SalvoVirga, 2016-03-07:
Within ROS we take leverage of rosdep to install all the required dependencies.
As in the wiki, you can run :

> `rosdep install --from-paths src --ignore-src -r -y`

from the root of your workspace and it will automatically install all the dependencies of the underlying modules.
Force each module, its dependencies are listed in the respective package.xml, rosdep automatically lookup for them. I imagine you might have the need to know all the dependencies in order to work within your grl framework, that would requires to go through all the package.xml files manually. 
That is a tedious job I would prefer not to do, that's why rosdep exists :P

ahundt, 2016-03-08:
There's just one file in grl listing all the dependencies: https://github.com/ahundt/grl/blob/master/BasisProject.cmake

Note that full list is for functionality well beyond kuka drivers.

I ended up being able to find all the components for iiwa_stack, clone and install them. It certainly wasn't automatic like you describe here, but perhaps I just missed the rosdep install instructions.

marcoesposito1988, 2016-03-11:
By the way, it is also possible to get the list of dependencies with the "rosdep keys" command, as in:
```rosdep keys iiwa_gazebo iiwa_hw iiwa_ros iiwa_msgs iiwa_moveit iiwa_control iiwa_description```
Salvo already mentioned the command needed to automatically install the dependencies. However, it can be that the package.xml files in the master branch are not up-to-date. I will create a virtual machine and check that all dependencies are installed by rosdep and the stack compiles without errors.

closed 2016-03-11 by marcoesposito1988