IFL-CAMP/iiwa_stack #4: Support for all iiwa_msgs?

ahundt, 2016-03-03:
Any messages besides JointPosition?

The readme said the app can "iiwa_ros_java: the ROSJava interface to use on SunriseApplications - it allows to send and receive messages defined in iiwa_msgs", however reading the actual ROSSmartServo code perhaps there are some changes implementing that functionality not publically available yet?

SalvoVirga, 2016-03-07:
Indeed it "can" send/receive any iiwa_msgs right away.
The ROSSmartServo applications was born as an exemplary application, it shows how to send and receive JointPosition messages. It contains instances of iiwaPublisher and iiwaSubscriber and they do the job, for example at line 135 `subscriber.getJointPosition()` it will give you the last commanded JointPosition. iiwaSubscriber has all the interfaces necessary to obtain any iiwa_msgs and do what you need to do with them. The method `publish()` of iiwaPublisher, will just publish any iiwa_msgs previously set (but only if a subscriber is listening).

In general, once you get a clear idea of the stack in the master branch I would suggest you to use the development one. There, ROSSmartServo already publishes the complete state of the robot and it allows robot control both with JointPosition and CartesianPosition (actually, iiwa_msgs is also different there), with a seamless transition between the two.
That branch has a lot of bug fixes and better functionalities than master, the downside is the nothing is documented in the wiki, but with some knowledge on ROS and the KUKA APIs it should be easy to use.

SalvoVirga, 2016-03-07:
More on that, we will probably merge the devel branch into master this week.

marcoesposito1988, 2016-03-11:
In the master branch it was possible to command the robot only in joint position mode. In development we added cartesian control, and we also switched back to standard ROS messages when a custom type is not necessary.

We will specify the possible control modes after merging development into master. Thank you for asking!

SalvoVirga, 2016-03-12:
New version of the framework is now in the master branch.
Also updated readme and wiki, hopefully they are more complete and clear now.

ahundt, 2016-03-14:
Interesting, thanks for the updates

closed 2016-06-19 by SalvoVirga