IFL-CAMP/iiwa_stack #5: kuka connectivity 1.9

ahundt, 2016-03-08:
there have been some changes in 1.9 so the iiwa_stack will need updates to match

SalvoVirga, 2016-03-08:
I am aware of that, moving to 1.9 means not being able to use the robot for some days for us, so we currently still use 1.7. 
We will probably move to 1.9 by end of next week, robot is busy until then. I might start to work on something meanwhile, but cannot test things. I'm already changing the wiki to clarify which KUKA Sunrise versions are supported.

SalvoVirga, 2016-03-08:
actually, I just remembered that in 1.8 they changed the namespace of their logger, that is fixed just modifying the imports. If that's the only thing also in 1.9, it's a 5 minutes job.

jeppewalther, 2016-05-23:
The SmartServo class seems to be missing. 

import com.kuka.roboticsAPI.motionModel.SmartServo

SalvoVirga, 2016-05-23:
Do you have the Connectivity module installed? Sometimes you get an installer for Sunrise Workbench 1.9 and a separate addon/zip file to install the Connectivity package.

jeppewalther, 2016-05-23:
Yes i have installed the FRI package i am assuming that is the same as the Connectivity package.

SalvoVirga, 2016-05-23:
You should also install the Servoing module, it should be something named "KUKA Sunrise.Servoing 1.9".

jeppewalther, 2016-05-23:
Ok, thankyou i seem to be missing this package

SalvoVirga, 2016-06-19:
There is a branch ready for Sunrise1.9 now, also cleared that in the wiki.
Master branch should move to 1.9 in the next days

closed 2016-06-19 by SalvoVirga