IFL-CAMP/iiwa_stack #6: Does iiwa support applications like gravity compensation？

lisikan, 2016-03-12:
Hi,thank you for your sharing. I have noted that the last version of kuka lbr supports the gravity compensation via FRI. In your work, position command is transferred via a variable defined in iiwa_msg, and KUKA FRI is not actually integrated. I don't  know if the new IIWA series interface includes that function. And if it does, how would you include the FRI interface functions in the future work? Any differences between the two versions of FRI?

SalvoVirga, 2016-03-12:
Hi,

I'm not sure at what you refer as "the last version of kuka lbr".
If you refer at the LBR 4+, its FRI library (Fast Research Interface) is to be considered completely different from the by IIWA's FRI (Fast Robot Interface), even if they share some features.
If you refer at the latest Sunrise Connectivity version (1.9), we didn't have the chance to use it yet and I don't remember if their plan was to add a graviry compensation feature to that. We will probably move to 1.9 by the end of the month, colleagues are using the robot right now and we don't want to risk moving to a new version and having something not working.

To my knowledge, KUKA allows gravity compensation for the IIWA only in with their "hand-guiding move", that requires a special flange/tool with hand-guiding enabling button. But again, no idea if they changed their mind with the 1.9 FRI.

From our side, we allow to turn on/off gravity compensation mode within our [ROSMonitor](https://github.com/SalvoVirga/iiwa_stack/blob/development/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSMonitor.java) application in the current [development branch].
I'm currently merging it into master and update the whole WIKI to work with it (I'm literally doing it right now). 

Regarding the integration with FRI, once we solve #7 we need to add FRI commands to our framework. It shouldn't be hard to have different applications using either SmartServo or FRI, but we would rather have a single application able to dynamically switch between the two. That might be tricky.

SalvoVirga, 2016-03-12:
Merged develop into master branch, also update readme and wiki.
[Here](https://github.com/SalvoVirga/iiwa_stack/wiki/rosmonitor) you can find how we support gravity compensation mode.

lisikan, 2016-03-13:
Thank you for answering. I reviewed your WIKI and  got your ideas. In fact, what I really care about is that whether the SmartServo joint impedance control could set an extra torque. I want to develop a Joint effort controller in ROS by just mirroring the current joint angle as commands and setting the damping and stiffness to zero. For LBR 4+, there is a function doJntImpedanceControl(JntPosition, newJntStiff, JntDamp, JntAddTorque, false). I don't know whether the iiwa can do this.

SalvoVirga, 2016-03-14:
Sure, that is the main idea behind our [ROSMonitor](https://github.com/SalvoVirga/iiwa_stack/wiki/rosmonitor) application. There for every joint we set their stiffness to 0 and damping to 0.7 (not to 0 or the robot will just fall) when the GravComp mode is activated with the physical button on the SmartPad.
You can also have the same behaviour modifying at runtime the joint impedance control from KUKA using our ROS Service, you can read about that [here](https://github.com/SalvoVirga/iiwa_stack/wiki/controlmode).
If it's clear and you don't have follow up question I would close this and put a link to it in the wiki's faq section.

closed 2016-03-14 by lisikan