IFL-CAMP/iiwa_stack #11: Gazebo simulation for LBR iiwa 7 R800

Ahyan1, 2016-03-18:
I am using a real robot LBR iiwa 7 R800. I also want to simulate this in Gazebo. However, it is stated that "Simulation with Gazebo is currently fully functioning ONLY for the KUKA LWR IIWA R820 (the 14 kg model).
The reason is that inertia values for the IIWA R800 model are not defined in its URDF yet."

I would like to know is there any way to simulate  LBR iiwa 7 R800 model using your package? Or when gazebo support for this model will be available? Also can I control my robot using MoveIt? Does your package iiwa_moveit support LBR iiwa 7 R800 model?

SalvoVirga, 2016-03-18:
So, the simulation of the 7kg version is only missing those inertia values for its links. You can see that [the robot description of the 14kg version](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_description/urdf/iiwa14.xacro) has already the inertia argument for the links, that was not made by us. We are not really Gazebo users, so it's always been a low priority issue for us.

We cannot use the real inertia values of the robot, so the idea would be to approximate them representing the robot links as cylinders and use the inertia of those. The same approach is also used for the Universal Robot package, for example in the [UR5 description](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_description/urdf/ur5.urdf.xacro). The use a macro name "*cylinder_inertial*" that just needs radius, length and mass of the cylinder that approximates a link.
If you feel like contributing with that yourself, it would be amazing. Else, I could try to work on it in the next days, but no promises!

About MoveIt, yes you can control both the real robot and its Gazebo simulation using it. The real robot example is [here](https://github.com/SalvoVirga/iiwa_stack/wiki/commandrobotmoveit). The Gazebo simulation just uses the same controllers used for the real one, so it would be pretty much the same.

daeunSong, 2016-04-15:
Hello, Is this answer of yours still valid?
If so, iiwa_7.urdf file should be missing some inertia values, but I see no values missing from the file. Also, when I look into iiwa_gazebo.launch file, it is uploading the iiwa_7 model, not 14, and model itself has no problem.

I am guessing that you have updated your iiwa_7.urdf file, but I just want to make sure. Am I right?
thank you.

SalvoVirga, 2016-04-15:
[iiwa_gazebo.launch](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_gazebo/launch/iiwa_gazebo.launch) loads by default the iiwa 14kg model
> Line 16 :     <arg name="model" default="iiwa14"/>

You can call the launch file as `roslaunch iiwa_gazebo.launch model:=iiwa7` in terminal or include it in another launch file adding the right arguments. Doing that it will load the 7kg model, but as I wrote there the inertial values for that are missing.
In the URDF of the 7kg there are some inertia values actually, but I believe they are just some copy/paste from the 14 model. But please, let me know if I'm wrong and that also works (that would be actually good).

SalvoVirga, 2016-06-19:
closing this, both model should work in position control, the issue about the inertia parameters is described in the wiki

closed 2016-06-19 by SalvoVirga