IFL-CAMP/iiwa_stack #13: Error with LBR IIWA: robot is not connected

Ahyan1, 2016-03-19:
I am trying to setup your packages to work with Kuka LBR IIWA. I am using ubuntu 14.04 LTS and ros Indigo. I found the robot connected with an wireless setting. Therefore, I didn't change the configuration. I uploaded the sunrise project to controller with no error or red sign in IDE. I followed roscore side. Next I ping with cabinet address and got reply. 
However, when I run the lunch file using  roslaunch iiwa_tool_examples iiwa_tool_read.launch, I found the error, [ERROR] [1458371985.836149274]: Robot is not connected.....  
 
afroza@afroza-XPS-8700:~/ros_ws$ roslaunch iiwa_tool_examples iiwa_tool_read.launch
... logging to /home/afroza/.ros/log/8b1ff4da-eda1-11e5-8d62-989096a52895/roslaunch-afroza-XPS-8700-6122.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://afroza-XPS-8700:41750/
SUMMARY
========
PARAMETERS
 * /iiwa/ReadRobotState/joint_position_topic: /iiwa/state/Joint...
 * /iiwa/ReadRobotState/ros_rate: 1000
 * /iiwa/publishJointStates: True
 * /iiwa/robot_description: <?xml version="1....
 * /iiwa/robot_description_kinematics/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /iiwa/robot_description_kinematics/manipulator/kinematics_solver_attempts: 3
 * /iiwa/robot_description_kinematics/manipulator/kinematics_solver_search_resolution: 0.005
 * /iiwa/robot_description_kinematics/manipulator/kinematics_solver_timeout: 0.005
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_1/has_acceleration_limits: False
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_1/has_velocity_limits: True
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_1/max_acceleration: 0
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_1/max_velocity: 10
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_2/has_acceleration_limits: False
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_2/has_velocity_limits: True
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_2/max_acceleration: 0
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_2/max_velocity: 10
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_3/has_acceleration_limits: False
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_3/has_velocity_limits: True
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_3/max_acceleration: 0
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_3/max_velocity: 10
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_4/has_acceleration_limits: False
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_4/has_velocity_limits: True
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_4/max_acceleration: 0
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_4/max_velocity: 10
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_5/has_acceleration_limits: False
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_5/has_velocity_limits: True
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_5/max_acceleration: 0
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_5/max_velocity: 10
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_6/has_acceleration_limits: False
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_6/has_velocity_limits: True
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_6/max_acceleration: 0
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_6/max_velocity: 10
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_7/has_acceleration_limits: False
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_7/has_velocity_limits: True
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_7/max_acceleration: 0
 * /iiwa/robot_description_planning/joint_limits/iiwa_joint_7/max_velocity: 10
 * /iiwa/robot_description_semantic: <?xml version="1....
 * /iiwa/toolName: tool
 * /rosdistro: indigo
 * /rosversion: 1.11.16
NODES
  /iiwa/
    ReadRobotState (iiwa_tool_examples/read_robot_state)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz_afroza_XPS_8700_6122_4091623036572324931 (rviz/rviz)
ROS_MASTER_URI=http://172.31.1.2:11311
core service [/rosout] found
process[iiwa/robot_state_publisher-1]: started with pid [6143]
process[iiwa/ReadRobotState-2]: started with pid [6144]
process[iiwa/rviz_afroza_XPS_8700_6122_4091623036572324931-3]: started with pid [6145]
[ERROR] [1458371905.833199520]: Robot is not connected...
[ INFO] [1458371905.886202985]: rviz version 1.11.10
[ INFO] [1458371905.886257469]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1458371906.552963147]: Stereo is NOT SUPPORTED
[ INFO] [1458371906.553024826]: OpenGl version: 4.5 (GLSL 4.5).
[ INFO] [1458371907.212131210]: Loading robot model 'iiwa7_tool'...
[ INFO] [1458371907.212188297]: No root joint specified. Assuming fixed joint
[ERROR] [1458371910.833342426]: Robot is not connected...
-------------
Please help me in this regard.
Thank you

marcoesposito1988, 2016-03-21:
Hi Ahyan,

that error message is normal until you start the ROSMonitor or ROSSmartServo application from the SmartPad of the robot. After you start the application, you should see the message "robot connected".

By the way: if I understand you correctly, you are using a WiFi network to connect the workstation to the robot. This is quite dangerous, given the unreliability of a wireless connection; as far as I know it would be much safer to use an Ethernet cable, better if without switch between the computer and the robot.

Ahyan1, 2016-03-21:
I started the application.However it shows that "couldn't find an AF_INET address for [fe80]. Robot is not connected...." .Do you think the problem occurs for using wifi network? Using wired connection can solved this?
Thank you

Ahyan1, 2016-03-23:
I tried by removing switch between IIWA and my PC and simply connect the cable with my pc, configure the IP address of associated network card. Using this network setting I am able to upload sunrise application on to the controller.
Then I started the ROSSmartServo application from the SmartPad and executed roscore.  I can hear the robot being unlocked. 
However, when I try t rosrun iiwa_tool_examples read_robot_state, It shows
"couldn't find an AF_INET address for [fe80]. Robot is not connected...."
roscore terminal also shows same error message "couldn't find an AF_INET address for [fe80]." 
Would you please give any suggestion to fix it?

Thank you

SalvoVirga, 2016-03-23:
You correctly use the KLI port to synchronize the Sunrise project onto the cabinet.
But the ROS communication goes through the KONI port at the moment, you need to set it up following [these instructions](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup).
You will still need the connection via the KLI port for synchronization (by default 172.31.1.147 is the cabinet and 172.31.1.2 your PC, from what you posted).
Then you are going to have another network for ROS via the KONI port. We use 160.69.69.69 on the cabinet and 160.69.69.100 on the PC, but you can choose whatever you want. You need to set those IPs : on the [network configuration within the Windows7 Embedded on the cabinet](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup), on [the network configuration of your PC and as your ROS_MASTER_URI](https://github.com/SalvoVirga/iiwa_stack/wiki/roscore_setup) and in the [configuration file in the Sunrise project](https://github.com/SalvoVirga/iiwa_stack/wiki/sunrise_project_setup). 
If your PC doesn't have two ethernet interfaces, you can simply switch the one it has after synchronizing the project.
I hope this helps, let us know in case it is not clear.

Ahyan1, 2016-03-23:
Thank you for your explanation. It helps me a lot. Before going for the cabinet setup (https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup), I would like to ask you (as there is a warning) if these instructions create any problem to the original setup of the cabinet/hardware, is it possible to restore the original setup?
Thank you

SalvoVirga, 2016-03-23:
An unofficial statement from KUKA would sound something like "That is not the intended use and it may lead to untested/unexpected behaviours". In practice we never experienced any problem at all, and it's possible to revert the change as shown in the note at the end of that wiki page. I personally assigned that ethernet port back and forth to win/rtos hundred times.

An idea to be super safe could be to make an image of the ssd disk inside the cabinet for backup.

Ahyan1, 2016-03-24:
Thanks a lot for your response. I didn't get any login information (ID/PW) of cabinet from Kuka service support. How can I obtain this information. Also, you mentioned that "There are at least 3 methods to have access to the Cabinet...". Would you please provide the pointers?
Thank you

Ahyan1, 2016-03-25:
I finished cabinet setup (https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup). Now IP addresses are:
cabinet: 192.168.168.2 (IP), 255.255.255.0 (netmask), 192.168.168.0 (gateway)
ubuntu machine: 192.168.168.1 (IP), 255.255.255.0 (netmask), 192.168.168.0 (gateway)
export ROS_MASTER_URI=http://192.168.168.1:11311
export ROS_IP=192.168.168.1
config.txt: master_ip=192.168.168.1

Then I started the ROSSmartServo application from the SmartPad and executed roscore. I can hear the robot being unlocked. 
However, when I try t rosrun iiwa_tool_examples read_robot_state, It shows
"Robot is not connected...."
 
Would you please give any suggestion to fix it?

Thank you

SalvoVirga, 2016-03-28:
The setup looks good. 
Run ROSSmartServo or ROSMonitor from the SmartPad and just a roscore on the external PC. Now try to print the joint position of the robot with
`rostopic echo /iiwa/state/JointPosition`
you should get some values, same for /iiwa/state/CartesianPose and so on. If this doesn't work let me know. Remember that **export ROS_MASTER_URI** and **export ROS_IP** has to be in the .bashrc file or you have to call that for every instance of the shell.

It might also be that those messages "*No LBR IIWA is connected. waiting for the robot to connect before reading*" are printed out (wrongly) also when the robot is connected, just check with rostopic echo. 

Mind that the examples from iiwa_tool is made to show how to work with a tool attached on the robot, as described [here](https://github.com/SalvoVirga/iiwa_stack/wiki/attachtool).
As shown at the bottom of that page, that requires the tool to be defined also within Sunrise Workbench. 
Rather than trying with that, I would call `roslaunch iiwa_moveit moveit_planning_execution.launch sim:=false`. That will make you work with the robot without any tool attached to it.
Then if you need to attach a tool, follow the wiki step by step to reproduce the examples of iiwa_tool for your specific tool. Remember to [define the tool you want to use in the launch file that you call](https://github.com/SalvoVirga/iiwa_stack/wiki/attachtool#selecting-a-tool).

Ahyan1, 2016-03-29:
Thank you for your kind response. I want to follow your suggestions. However, now I am facing another new problem that didn't happened earlier. Before I was able to execute ROSSmartServo or ROSMonitor from smart pad while running roscore on ubuntu machine without any red error messages.  However, now it shows the following errors on smartpad
[image]
Do you have any idea why this error?
Thank you

SalvoVirga, 2016-03-29:
Oh, which Sunrise version are you using? I had this error an infinite number of times in the past, but it should be fixed by now.
Just restart completely the cabinet until it doesn't show up :|
When it shows up you could also create a diagnostic package (check in the KUKA manual) and send it to me, I would then forward it to the right people at KUKA.

Ahyan1, 2016-03-29:
I am using sunrise.os 1.7.0.11.  How did you fixed it when you faced this error? 
Here is the diagnostic package.
https://drive.google.com/file/d/0B8X1OCrk2KDvR0lwbDNKcXhUOUU/view?usp=drive_web
Thank you

SalvoVirga, 2016-03-29:
Just turning off/on the cabinet until you are lucky :|

Ahyan1, 2016-03-30:
Now all error messages are disappears. Again I am able to start the ROSSmartServo application from the SmartPad and executed roscore .
[image]
I can hear the robot being unlocked. Then I use rostopic list. It shows:
/iiwa/command/CartesianPose
/iiwa/command/CartesianWrench
/iiwa/command/JointPosition
/iiwa/command/JointStiffness
/iiwa/commmand/JointTorque
/iiwa/joint_states
/iiwa/state/CartesianPose
/iiwa/state/CartesianWrench
/iiwa/state/JointDamping
/iiwa/state/JointPosition
/iiwa/state/JointStiffness
/iiwa/state/JointTorque
/iiwa/state/buttonEvent
/rosout
/rosout_agg
Then I try to print the joint position of the robot with rostopic echo /iiwa/state/JointPosition. But terminal shows nothing. Same for /iiwa/state/CartesianPose.  
Next I call roslaunch iiwa_moveit moveit_planning_execution.launch sim:=false. It keep showing that "No LBR IIWA is connected. waiting for the robot to connect before reading". Also it shows visualization tool. Now if I execute rostopic echo /iiwa/joint_states, it shows following response:

Although other (e.g. rostopic echo /iiwa/state/JointPosition) show nothing.
I put export ROS_MASTER_URI and export ROS_IP in the .bashrc file. I wonder is the robot really connected. Why it is not showing any values for rostopic echo?
Hope you will give me suggestions to proceed on

Ahyan1, 2016-03-30:
I am using sunrise.os 1.7.0.11.
I have created a diagnostic package, but the zip file is too big (79.4 MB). How to send this?

closed 2016-05-21 by SalvoVirga

PatrikBystricky, 2016-12-08:
I try to plan the movements of the robot Kuka LBR IIWA. I am using ubuntu 14.04 LTS and ros Indigo. But the robot writes an error: No LBR IIWA is connected. Waiting for the robot to connect before reading...
What am I doing wrong, if I want to plan the robot movements? Thanks.
[image]

[image]

When you choose Planning / Commands / Plan and Execute robot does not move, only writes an error in Terminal.

[image]

Thank you.

SalvoVirga, 2016-12-08:
That "No LBR IIWA is connected. Waiting for the robot to connect before reading..." is quite buggy, it is printed also when the robot is connected. So don't rely on that, we will fix it at some point.

If you are running ROSSmartServo and you can read the state of the robot, you should also be able to command it. I think your problems falls in the category of network setting, [please have a look at this.](https://github.com/SalvoVirga/iiwa_stack/wiki/FAQ#i-can-read-the-state-of-the-robot-but-it-doesnt-move-when-i-send-commands)
In particular, be sure you set everything in the ~/.bashrc. Once you check the three points, let me know if you find something or nothing changes.

PatrikBystricky, 2016-12-08:
I went through this again three fundamental steps. When I started ROSSmartServo so after starting the program the robot a click and after 5-6 seconds the stop Ros loop. I know Ping device from both sides. ROS - Cabinet and Cabinet - ROS.  I still have new problems. I can not get them out. Thank you for your advice.
[image]
[image]
[image]
[image]

[image]

jacknlliu, 2018-04-23:
I add a line about sleeping 1ms to [ROSBaseApplication.java](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSBaseApplication.java#L222), and it solved this issue.

```
import com.kuka.common.ThreadUtil;

			while(running) { 
				decimationCounter++;

				// This will publish the current robot state on the various ROS topics.
				publisher.publishCurrentState(robot, motion, toolFrame);
                               
                              // sleep 1ms for avoiding "Update of SPR did not succeed within 1 second"
                               ThreadUtil.milliSleep(1);

				if ((decimationCounter % controlDecimation) == 0)
					controlLoop();  // Perform control loop specified by subclass
			} 
```

PR #17 (2016-05-21): ntp_with_host set to false
default true was causing the java application to hang without a ntp server on the linux side.
Fixes #13
commit 85b5da4d (2016-05-21): Merge pull request #17 from SalvoVirga/hotfix/noNTP

ntp_with_host set to false. Fixes #13