IFL-CAMP/iiwa_stack #18: Subscribers running on robot is never executed

jeppewalther, 2016-06-08:
I am running Sunrise 1.9 Java 1.6 on the KUKA controller and ROS Indigo on the control PC. 

The robot seems to be publishing info just fine. However i am not able to subscribe to subjects, say "/iiwa/command/JointPosition". 

I have also tried to setup a simple /Float64 subscriber with the same results. It seems that the message listener (onNewMessage) callback is never executed. 

I can see the ROSJava node is registred in the ros graph and that the topic that i am subscribing to gets registered. But again for some reason the subscribers ROSJava callback never gets executed

I saw in the documentation that you are using Java 1.7. In that case how did you get 1.7 installed on the controller.

SalvoVirga, 2016-06-08:
Hi, iiwa_stack should work as well with Sunrise 1.9, I'm switching myself to that version in the next weeks. Downgrading to 1.7 requires assistance from KUKA, so I don't suggest it.

About your problem, did you modify your .bashrc file according to [this](https://github.com/SalvoVirga/iiwa_stack/wiki/roscore_setup)?

jeppewalther, 2016-06-09:
Yes both ROS_IP and  ROS_MASTER_URI are set. 

With my simple publisher subscriber setup the funny thing is that my listener "/iiwa/listner" seems to get connected just fine to the java node "/iiwa/iiwa_subscriber" when displaying "Nodes/Topics (all)"  but when displaying "Nodes/Topics (active)" the connection is not displayed. 

Also i am able to publish info from the robot to the ROS pc

SalvoVirga, 2016-06-14:
Problem is fixed for the user. Usually this kind of problems come from some mismatching network settings, if happens always check : 
- [the IP set to the KONI interface within Windows7 Embedded](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup), 
- [that the .bashrc contains the right IPs](https://github.com/SalvoVirga/iiwa_stack/wiki/roscore_setup),
- [the config.txt in Sunrise Workbench has the right IP](https://github.com/SalvoVirga/iiwa_stack/wiki/sunrise_project_setup#note-for-developers)

closed 2016-06-14 by SalvoVirga