IFL-CAMP/iiwa_stack #20: sunrise/ROS time offset

Haas11, 2016-06-28:
I was having difficulties visualizing the robot in rviz. 
Some digging showed that the robot_state_publisher is giving timing errors on the tf's.
Turned out the smartpad clock is an hour ahead of the (correct) system time on the ROS side. 
Probably some daylight saving issue. Any idea how to sync these timers? sunrise Doc turns out empty.
Bug could be easily catched by checking if offset is around 3600 seconds.

SalvoVirga, 2016-06-28:
Indeed, we had the same problem some time ago. We couldn't find a way to configure the time on the SmartPad, changing the time from the Windows7 Embedded system doesn't do the job. We report bugs/errors to KUKA on a regular basis, I believe we already pointed that out to them.

Since we had the problem, we provide the solution!
What you need is to synchronize the two systems' clock using a Network Time Protocol (NTP) server, simply install an NTP server on the Linux machine and configure it to work on the KONI subnet. In the [config.txt](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/config.txt) file on the Sunrise/Java side there is a flag to use a NTP server, set it to true. Run the ROSMonitor or ROSSmartServo and the times will be synchronized, they take few seconds (maybe less) to converge. If you set it to true and you don't have a NTP server on Linux, the Java application will hang doing nothing. On the SmartPad you should get a notification of which TimeProvider is the application using.
Ask more if you need, @marcoesposito1988 knows better but he is away this week.

Be aware that with the current implementation, using the NTP functionality reduces a bit the performances of the Java application. But you won't probably see the difference, we discovered that recently looking for something else, we will soon release a new refactoring of iiwa_stack that takes care, also, of this.

Haas11, 2016-06-29:
Thank you so much! 
Indeed an NTP server solves this issue.

closed 2016-06-30 by SalvoVirga

ahundt, 2016-07-05:
Hey this sounds like it will be relevant to me, could @Haas11 could you maybe add what you did to the wiki? Did you connect to windows7 with a keyboard/mouse to set up the NTP server or can it be done from Java? 

Thanks!

SalvoVirga, 2016-07-05:
Within iiwa_stack is all done on the Java side, you just need an NTP server on the Linux machine (configured to work on the right subnet) and to set the flag to true in that [config.txt](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/config.txt) file. Then our Java code takes care of everything.

Haas11, 2016-07-06:
Yes. There's a standard ntp application in the Ubuntu repository. Just install through terminal, Modify the configuration file of that app to broadcast it's time to the appropriate ethernet port. 
Then set the flag true in config.text, deploy through workbench and it should indicate it on the SmartPad.

ahundt, 2016-07-12:
cool, thanks!