IFL-CAMP/iiwa_stack #21: Incorrect frame for    ServoMotion.validateForImpedanceMode

Haas11, 2016-06-30:
I noticed when switching control mode to Cartesian Impedance, load mass validation is performed on line 235 in:  
_"iiwa_stack/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSSmartServo.java"_
This is done on the frame: "LBR_iiwa_7_R800_1", even when an additional tool is defined.
(The selected tool has a definition and attached frame defined in sunrise and the smartpad indicates it is selected when starting roscore)
This then often results in a variance>10 and a subsequent failure to initiate the control mode.

How to tell sunrise it should use the currently selected tool frame for load mass validation? 
It seems like robot is initialized correctly at line 214: 
_robot = getContext().getDeviceFromType(LBR.class);_
but apparently not entirely

SalvoVirga, 2016-06-30:
I just noticed that the master branch is more behind that I thought.
Have a look at [this branch](SalvoVirga/iiwa_stack/tree/feature/ConstantForce) for the moment. That has the corrent calls to properly work with a tool once you have the rosparam defined (i.e. it publishes the cartesian pose of the tool and it uses the tool frame for commands). I also modified the validateForImpedanceMode call to be done for the robot+tool and not just the robot.

I will try to merge that branch at least to development during the weekend.
Take take of two things if using that branch : 
- update your ROSJavaLib/iiwa_msgs-1.5.0.jar with the one in that branch, just that one, it has some difference with the one in master
- that branch is still for Sunrise1.7, if you use 1.9 you need to modify a couple of imports, just delete the lines marked in red within in imports and put the new ones.

Haas11, 2016-06-30:
Indeed also some other issues when working with tools. I'll clone the other branch for now. Thanks!

SalvoVirga, 2016-06-30:
let me know if you find anything wrong also in that one

SalvoVirga, 2016-07-01:
I just noticed that I made some changes and now the CartesianPose from the robot is correct but it uses the tool frame as frame_id, while all poses are still in robot space so I should use the base frame.
I will change that quickly tomorrow on the ConstantForce branch, then we will probably focus on pushing the new factoring into development or even master as soon as possible. That already includes these changes.

mohissa55, 2016-07-02:
@SalvoVirga I cannot access "this branch" mentioned above. Can you please make it available because I am having the same issue with variance limits. 
+ 
I am using the master branch. Do you suggest switching to another one or is it the most updated version?

SalvoVirga, 2016-07-02:
@mohissa55, the branch is in this repo, [here](https://github.com/SalvoVirga/iiwa_stack/tree/feature/ConstantForce).
You can just clone it :
`git clone -b feature/ConstantForce https://github.com/SalvoVirga/iiwa_stack.git`

or checkout that branch from your local git repo :
`git fetch`
`git checkout feature/ConstantForce`

Please mind the remarks I wrote above, I didn't have much time to work on it yet.

SalvoVirga, 2016-07-02:
Anyway, be aware that the error with "_variance>10_" also shows up in other occasions, e.g. if the mass of the tool is not properly set, if there is an external force applied on the robot while switching the dynamic mode, if the robot is in a singularity position while switching.
So it is not necessarily due to the implementation.

SalvoVirga, 2016-07-29:
fixed with #30, now in branch development

closed 2016-07-29 by SalvoVirga