IFL-CAMP/iiwa_stack #25: SmartServo Motion's relative velocity has no effect

Haas11, 2016-07-14:
Setting a relative (joint) velocity does not seem to make any difference when sending Cartesian pose commands in Cartesian Impedance mode. 
Any value in <0,1] gives the same resulting motion for me. 
Although seems to be properly defined on java side:

`public void configureSmartServoMotion(iiwa_msgs.SmartServoMode ssm, SmartServo mot) {
		if (mot == null)
			return; // TODO: exception?

		if (ssm.getRelativeVelocity() > 0)
			mot.setJointVelocityRel(ssm.getRelativeVelocity());
		mot.setMode(buildMotionControlMode(ssm));
}`

SalvoVirga, 2016-07-14:
Ye, we also noticed that some time ago but haven't time to dig into it.
Have you also tried to set the mode of the service request to -1?
If you give -1, it should use the same control mode of the last request but just change the speed, see [here](https://github.com/SalvoVirga/iiwa_stack/blob/feature/ConstantForce/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSSmartServo.java#L249-250).
It might also not work like that, I think we already tried. In that case, the problem could be that the velocity can only be set before the motion starts and not at runtime, that is what we do [here](https://github.com/SalvoVirga/iiwa_stack/blob/feature/ConstantForce/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSSmartServo.java#L334). So we might need to discard the current motion and make a new one just for that...quite annoying (probably we already reported that to kuka)

Haas11, 2016-08-17:
For the (safety) record:
Being in Cartesian Impedance mode, setting the mode scalar to -1 makes it revert back to a position control scheme.
At least for the branch I am currently on (constantForce), which I think by now is the development branch.

SalvoVirga, 2017-02-02:
Fixed in release 1.2: 943299a

closed 2017-02-02 by SalvoVirga