IFL-CAMP/iiwa_stack #26: Destination reached flag

Haas11, 2016-07-14:
Is there already a way of polling whether a destination has been reached?
The KUKA servoing manual states:
> 
The IServoRuntime class contains the isDestinationReached() method. 

> Notification upon achievement of target:
For SmartServo and SmartServoLIN motions, the listener SampleGoal-
ReachedEventListener is available. It signals that the destination setting has
been reached using the event onGoalReachedEvent(...).
The listener must be derived from the class IServoOnGoalReachedEvent and
be registered on the interface IServoRuntime by calling the method setGoal-
ReachedEventHandler(...).

Could be a handy feature for on the ROS side.

SalvoVirga, 2016-07-14:
Yes! It is a nice feature that we plan to add soon, we are still not sure if we want to just have a topic for that or to use an actionlib to get both notifcation for a goal being reached and feedback on the current progress.
Maybe it is a good idea for us to make a public roadmap page in the wiki, so that others know what we plan for. :D 
And anyway, any contribution or suggestion is always welcome!

Haas11, 2016-07-14:
a public roadmap might be a good idea indeed. 
I'm using these packages in combination with the new ROS functionality in MATLAB/Simulink.
(Although not ideal for RT comms).
Have not yet really dived into the java side of things because I'm not the best java coder and the graduation deadline is fast approaching ;) So right now I'm mainly running into implementation issues.
But I'll keep sharing things as they come up. Can perhaps share some templates for functioning examples of applications that include MATLAB & Simulink.

In any case thanks for all the good work!

SalvoVirga, 2017-02-02:
There is now a service that does exactly that job! Thanks for the good idea!

closed 2017-02-02 by SalvoVirga