IFL-CAMP/iiwa_stack #28: Does the interface support torque?

sinamr66, 2016-07-21:
Hi All,

I have a quick question. Does the interface support torque? The interface has a massage type dedicated to torque. Does it work? and another minor question,  can we get the robot's mass matrix?

Thx a lot
Best
Sina

Haas11, 2016-07-21:
The java side has functionality to overlay either a joint torque or Cartesian force. 
Not sure if already working through ROS. There's a branch dedicated to its development. 

The dynamic model of the robot is hidden, both symbolically and numerically.

sinamr66, 2016-07-21:
Thx a lot for the reply

SalvoVirga, 2016-07-24:
Torque control "should" be available from KUKA via their Fast Robot Interface (FRI), I personally never tried it nor know anyone who did. In general the FRI interface is quite obscure and we don't feel the need to use it. In theory you should be able to have joint state monitoring and joint or torque control at 1kHz with it, but again, we never needed it and we have some doubts about it, so there is no ROS interface for it so far. Also, we use the Ethernet port that KUKA uses for FRI for other things, so at the moment the two things cannot live together.
In issue #7, @Haas11 recently showed that is possible to use a USB/Ethernet adapter on the cabinet instead of the KONI port. That might open the way to also integrate an interface to FRI.

About the robot's mass matrix, it is not publicly available, so nope.

closed 2016-12-12 by marcoesposito1988