IFL-CAMP/iiwa_stack #31: Nullspace dynamics

Haas11, 2016-08-25:
I'm having some trouble with the elbow dropping down and into a limit when in cart imp mode. 

To my mere human judgement, a stiffness and damping (in nullspace) of 5 and 0.1 feel practically the same as a stiffness of 200 and a damping ratio of 0.7. 

I know the end-effector mass specs to be slightly off since it is so light, but it was workable before. The nullspace properties are set correctly in ROSSmartservo.Java as far as I can tell. 

Anyone else noticed nothing changing in nullspace dynamics when defining them upon a service request? Or a way of preventing elbow drop? endpoint dynamics do not seem to affect it.

marcoesposito1988, 2016-12-12:
We are not using this currently, but my wild guess would be that you need to pass a suggested joint position along with the desired cartesian position in order to have nullspace stiffness working. I do not know if the controller will just use the current joint position otherwise.

Have you made any progress on this?

SalvoVirga, 2017-04-03:
I'm closing this since we had no further feedback. Feel free to open it again if you need more help.

closed 2017-04-03 by SalvoVirga