IFL-CAMP/iiwa_stack #34: Problems using Move Group Commands

mat-l, 2016-09-22:
Hello all,
I am trying to control an simulated kuka lbr iiwa robot with a cpp script using move group commands. 
The original script is not from me but a tutorial using an ur5 robot.
Anyway, after executing the script everything works fine but the robot doesn't move (with ur5 the similar script worked).
Is it possible that I need to change some names or publish to a different node than with the UR5?
If someone already created a program to control this robot using move group commands it would be really helpful if I could use that script as example to make mine work.
Thank you for your help.

[iiwa_script.docx](https://github.com/SalvoVirga/iiwa_stack/files/487932/iiwa_script.docx)
[iiwa_launch.docx](https://github.com/SalvoVirga/iiwa_stack/files/487933/iiwa_launch.docx)

marcoesposito1988, 2016-09-24:
Hi @mat-l,

probably that is because we nested the whole MoveIt nodes, topics and parameters into a namespace (by default "/iiwa") to make it possible to start multiple instances under a single roscore.

You do not need to modify the launch file (your iiwa_launch.docx). You can just use our moveit_planning_execution.launch, which resides into iiwa_moveit. Then you can use your C++ script unmodified (you will have to start it in the "/iiwa" namespace, though).

mat-l, 2016-09-29:
Thank you for your answer.
Could you explain to me what I have to do with a little more details?
I started the simulation with your moveit_planning_execution.launch command and afterwards used rosrun to start my cpp script. From that I get an Error message because "manipulator" could not be found. 
I suspect that is because I don't start my script in the /iiwa namespace properly. How do I start the script in this specific namespace? 
To use a Terminal in which the iiwa workspace is sourced seems to be not enough.

SalvoVirga, 2016-10-04:
Did you manage to solve this? You should just start your node within the right namespace (iiwa), from a roslaunch file it's pretty easy. Check the *ns* attribute [here](http://wiki.ros.org/roslaunch/XML/node)

mat-l, 2016-10-12:
Hey
thanks to your comment I now understand a little more about roslaunch and I was able to solve the communication problem. For now all I could do is to add something like a table into the scene and create error messages for planning.

[ERROR] [1476274928.891292938, 42.126000000]: RRTConnect: Unable to sample any valid states for goal tree
and 

[ERROR] [1476274918.882453812, 32.138000000]: Unable to transform from frame 'base_link' to frame '/world'. Returning identity.

I remember a similar problem that has been solved by adding a fixed joint between base_link and the world frame but I didn't had a chance to look into it yet.

marcoesposito1988, 2016-12-12:
Hi @mat-l,

sorry for the long absence. Did you make any progress with this?

SalvoVirga, 2017-04-03:
I'm closing this since we had no further feedback. Feel free to open it again if you need more help.

closed 2017-04-03 by SalvoVirga

mf093087, 2022-01-19:
I have the same error. 

[ERROR] [1642615719.527758477]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
[ INFO] [1642615719.529822789]: Loading robot model 'iiwa7'...
[ INFO] [1642615719.530232672]: No root/virtual joint specified in SRDF. Assuming fixed joint
[FATAL] [1642615719.784914838, 186.460000000]: Group 'manipulator' was not found.
Traceback (most recent call last):
  File "/home/user/catkin_stackiiwa/src/iiwa_stack/iiwa_ros/scripts/pythonmove.py", line 132, in <module>
    MoveItCartesianDemo()
  File "/home/user/catkin_stackiiwa/src/iiwa_stack/iiwa_ros/scripts/pythonmove.py", line 20, in __init__
    arm = MoveGroupCommander('manipulator')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 66, in __init__
    name, robot_description, ns, wait_for_servers
RuntimeError: Group 'manipulator' was not found.

srwhite1995, 2022-06-23:
Hi @mf093087, did you ever fix this? I am getting the same error. Many thanks, Sam.