IFL-CAMP/iiwa_stack #36: Support of Sunrise.OS 1.11

gekoraus, 2016-11-25:
As I need some features of Sunrise.OS 1.11, I was wondering if you are planning to support this version in the future?
In the FAQ is written, that normally only tiny changes are needed to support newer versions, so I'll probably also try for myself if I can get it to work.

SalvoVirga, 2016-11-25:
I got Sunrise.OS 1.11 few days ago. From what I've quickly seen, only a small change in [ROSSmartServo.java](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSSmartServo.java#L304-L309).
Lines 304-309 should become something like : 

```
case CARTESIAN_POSE: {
     PoseStamped commandPosition = subscriber.getCartesianPose();
     Transformation tr = helper.rosPoseToKukaTransformation(commandPosition.getPose());
     Frame destinationFrame = new Frame(tr);
     if (robot.isReadyToMove())
         motion.getRuntime().setDestination(destinationFrame);
}
```
But I didn't have the time to test 1.11 on the real robot. This should be the only change needed.
Let me know if everything is fine for you, I will probably install 1.11 on my robot later next week.

SalvoVirga, 2016-11-30:
branch [/feature/Sunrise1.11](https://github.com/SalvoVirga/iiwa_stack/tree/feature/Sunrise1.11) now supports the correspondent version, soon to be merged in master

closed 2016-11-30 by SalvoVirga

gekoraus, 2016-12-02:
Thank you very much! I just updated and everything is working as expected!

PR #37 (2016-11-30): Feature/sunrise1.11
Small modification according to the changes in the SmartServoRuntime.setDestination() interface.
Fixes #36.