IFL-CAMP/iiwa_stack #38: reason for 'static configuration'?

gavanderhoorn, 2016-12-06:
The current implementation of `iiwaConfiguration` has most of its member variables declared as `static` (the `staticConfigurationSuccessful` variable fi).

I was wondering if one of the authors (@marcoesposito1988, @SalvoVirga) could provide some rationale as to why this class was set up this way?

marcoesposito1988, 2016-12-07:
Hi @gavanderhoorn,

I guess this wanted to make the class behave like a singleton (it is instantiated only once in our code, but we would like to make it less error-prone for users who would wish to customize the applications heavily). However, it might be that some things can be queried right away, while others only after connection to the ROS master (stuff that is read from the config file in the Sunrise project vs ROS parameters).

It would be probably better to clean it up and make it a proper singleton, with a lazy instantiation. I will take a look at it ASAP, but suggestions are welcome! ;)

gavanderhoorn, 2016-12-08:
Thanks for the explanation. I figured it was on its way to become a proper singleton.

closed 2016-12-08 by gavanderhoorn