IFL-CAMP/iiwa_stack #41: question: using NtpTimeProvider.startPeriodicUpdates(..) instead of polling in ROSBaseApplication.run()

gavanderhoorn, 2016-12-08:
As per subject really.

I was wondering why `ROSBaseApplication.run()` invokes `NtpTimeProvider.updateTime()` instead of configuring the `NtpTimeProvider` instance to manage its own periodic updates using `NtpTimeProvider.startPeriodicUpdates(..)`.

Are the periodic updates too slow / inaccurate for the driver? It would seem to make it possible to do away with the decimation counters and make the while-loop tighter.

marcoesposito1988, 2016-12-08:
Very good observation! We should see how nice it plays with the threading in the Sunrise JVM. Worth to try!

Marco Esposito
Research Assistant

Computer Aided Medical Procedures (CAMP)
Technische Universit?t M?nchen
Institut fuer Informatik I16
Boltzmannstr. 3<x-apple-data-detectors://4/0>
85748 Garching<x-apple-data-detectors://4/0> b. Muenchen, Germany
campar.in.tum.de<http://campar.in.tum.de/>

E-Mail: marco.esposito@tum.de<mailto:marco.esposito@tum.de>
Phone: +49 89 289 19403<tel:+49%2089%20289%2019403>

gavanderhoorn, 2016-12-08:
I've tested it with `ROSMonitor` and that seemed fine. Will have to test it with actual motions, but I see no real difference between polling in the while and letting `*Executor` do it for us (other than a few extra context switches now and then): it basically calls the same code.

SalvoVirga, 2017-02-21:
Fixed, thanks for pointing that out!

commit e466995b (2017-02-21): Automatic periodic NTP updates

Fixes #41
PR #56 (2017-02-21): NTP Fix + small fixes
Fixes #41
closed 2017-02-21 by SalvoVirga