IFL-CAMP/iiwa_stack #43: Controlling two IIWA over the interface

sinamr66, 2017-01-06:
Hi All,

If I understand is correctly, the roscore is running on the cabinet (the control box). If it is true, is it possible to control two IIWA at the same time by using this interface? What are going to be the IPs, i.e. ROS_MASTER_URI, ROS_IP and the cabinet's IPs?

Thanks a lot for you help.

Best,
Sina

SalvoVirga, 2017-01-08:
Roscore runs on the Linux machine that one connects to the robot. What runs directly on the cabinet are ROS(Java) nodes that publish the current state of the robot and subscribe to commands. Therefore, the easiest way to use two IIWAs would be to connect both to the same machine where the roscore runs. 
Taking for example our current setup, the Linux machine will have the IP 160.69.69.100 (that would be [ROS_MASTER and ROS_IP to set on the Linux machine](https://github.com/SalvoVirga/iiwa_stack/wiki/roscore_setup)), IIWA_1 has the IP 160.69.69.69 and IIWA_2 would be 160.69.69.70. An Ethernet cable from each KONI port of the two IIWAs to the Linux machine, on both IIWAs the setup described in [here](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup).

Mind that in general, the two robot must have different names set in their [config.txt file](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/config.txt). The ROSJava nodes will create publishers and subscribers under a namespace accordingly to that given name. By default the topics are iiwa/state/... and iiwa/command/..., so if you change the robot_name field into  "luke" and "leia" for example, you will have luke/state/... and leia/state/... for the state topics of the two robots.

This would allow one to already have the state and command topics and use them, you don't **need** anything extra.     
However, you have to be sure that your ROS nodes/params/etc are created and work on the right workspace. 
For example, we have an argument "robot_name" in many launch files, that makes sure that everything is in the namespace "iiwa" by default, the same name as in the config.txt. You need to adapt that to your changes, maybe have a look on how that variable moves around starting from [the main launch file in our MoveIt! package](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_moveit/launch/moveit_planning_execution.launch)

I hope I was clear enough, if there are more doubts just let me know!!

sinamr66, 2017-01-08:
You were, Thank a lot :)...The description is very helpful..:)
I will test it and come back to you if face any problem which I can not figure it out...:)

closed 2017-01-08 by sinamr66