IFL-CAMP/iiwa_stack #44: Robot is not connected...

mat-l, 2017-01-06:
Hello all,
we want to control a real kuka iiwa robot using the iiwa_hw part of this package.
As you can see in the attached picture first we got the error : [Robot is not connected.]
and after that [Couldn't find an AF_INET adress for [[fe80]  ].
We already tried the solution you find with google like changing the EXPORT statement in .bashrc from localhost to the actual IP but that did not solve anything.
We think it could be related to the IPv6 since the kuka handheld outputs something like fe80:0:0:0:0%efe:c0a8:1 as the robot IP. 
We are using the kuka sunrise version 1.9.1.7.
[image]
[image]

Thanks for your help.

SalvoVirga, 2017-01-08:
The "Robot is not connected..." error is a stupid bug that I always don't bother to fix, it is printed also when the robot is actually connected, so please don't rely on that. I will fix it asap.

About the rest, did you perform the setup describe in [here](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup) on the robot cabinet?  
ROSJava at some point looks for available network interfaces to connect to a ROS Master, with the setup we describe we basically make the network interface on the KONI port available in Java (and therefore for ROSJava), so that it is found and used for our communication.     
Without the above setup ROSJava will only find the internal virtual network to the Real Time OS on the cabinet and try to use that, I think that is that weird IP address. 

If you did perform that setup tho, then let me know and I will think in some other directions!

mat-l, 2017-01-10:
Hi @SalvoVirga thanks for your relpy. We tried both with KONI interface and also using the standard X1 Ethernet plug on the front side of the KUKA iiwa robot. Nevertheless we were not able to execute and control the robot using ROS.
Regards

SalvoVirga, 2017-01-16:
I still didn't get if you perform the setup on the cabinet described [here](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup). Did you also get printer out that IP address with that setup?

mat-l, 2017-01-17:
Hey
Yes we performed the setup on the cabinet as well as the linux machine.
In the pictures you see the output generated after starting SmartServo and launching iiwa_tool_command_moveit.launch
We also tried to get some information with rosrun iiwa_tool_examples read_robot_state but the result is a Robot is not connected...Error (I know just ignore) and in the roscore Terminal it prints Couldn't find an AF_INET adress for [[fe80] (same as for iiwa_tool_command).
 

The IP Adresses are:
Linux Computer Network: 172.31.1.148
bashrc:
export ROS_MASTER_URI=http://172.31.1.148:11311
export ROS_IP=172.31.1.148
Cabinet: 172.31.1.147

[image]
[image]
[image]
[image]

Thank you for your support!

SalvoVirga, 2017-01-18:
Wait, if you did this [setup](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup), you should have now two available ethernet adapters on the cabinet.

- One to sync the Sunrise project (let's call it **SUNRISE Ethernet**) - **bottom right** of the front panel of the cabinet (right above the power button).
- One to be used with iiwa_stack (let's call it **KONI**), on the **bottom left** of the front panel of the cabinet (next to the usb ports).

This:
> The IP Adresses are:
> Linux Computer Network: 172.31.1.148
> bashrc:
> export ROS_MASTER_URI=http://172.31.1.148:11311
> export ROS_IP=172.31.1.148
> Cabinet: 172.31.1.147

looks weird to me.
**172.31.1.147** is the default IP for the **SUNRISE Ethernet** port, while you should use the **KONI** port for iiwa_stack.
You should have one ethernet cable from your computer to the **KONI** port when using iiwa_stack on another subnet (not 172.31.1.xxx). In the [setup](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup), for example, we use 160.69.69.69 for the robot and [160.69.69.100](https://github.com/SalvoVirga/iiwa_stack/wiki/roscore_setup) for the Linux computer.
You should use the interface on 172.31.1.xxx only to synchronise the Sunrise project on the robot. (Ideally you have two ethernet interfaces on the Linux machine and keep two Ethernet cables the whole time on).

So be sure that you have the right setup, you should be able to ping the robot from the Linux machine to its **KONI** interface (e.g. **ping 160.69.69.69** if you use the same IP from the setup page).

In general, [this FAQ](https://github.com/SalvoVirga/iiwa_stack/wiki/FAQ#i-can-read-the-state-of-the-robot-but-it-doesnt-move-when-i-send-commands) contains everything you need to check:

The final configuration will looks something like this:

Sunrise interface IP Cabinet: 172.31.1.147
Sunrise interface IP Linux Computer:  172.31.1.147

KONI interface Cabinet: 160.69.69.69
KONI interface Linux Computer: 160.69.69.100
bashrc:
export ROS_MASTER_URI=http://160.69.69.100:11311
export ROS_IP=160.69.69.100
in config.txt in the Sunrise project:
master_ip: 160.69.69.100
master_port: 11311

If only have 1 Ethernet interface on your Linux Computer, you need to switch between the two when synchronising and when using iiwa_stack. The subnet 160.69.69.xxx is just an example, you could use any other (not 172.31.1.xxx).

I hope this can help you and that I was clear, please write if not or if you encounter any other problem!

mat-l, 2017-01-18:
We adapted our Network settings to your suggestions.
The fe[[80]... Error is gone now. 
We still can't control the robot and there are a lot of Error Messages coming on the Touchpanel. (see pictures below.)
[image]
[image]
[image]

SalvoVirga, 2017-01-18:
it looks better! 
Can you ping the robot at 160.69.69.69 from the Linux Machine?
Which version of Sunrise do you have?
Which branch or commit of iiwa_stack do you use?

I remember that error was showing up when I was moving from Sunrise 1.7 to Sunrise 1.9. If you use Sunrise1.9 you should use [this branch](https://github.com/SalvoVirga/iiwa_stack/tree/master-Sunrise1.9) (`git clone -b master-Sunrise1.9 https://github.com/SalvoVirga/iiwa_stack.git`)

mat-l, 2017-01-20:
Yes the ping works.
We use the most recent version. That should be 1.9 .
We used the older Version. After the update to the branch you recommend we got error messages again. But that was only for the first try. Later today we will dig into it and start with reloading the java_ros part using the file in the new branch.

[image]

SalvoVirga, 2017-01-20:
Yes, be careful with the jar files in ROSJavaLib, they change between branches.
Delete all the old one and add the new ones, if you have errors because some required jar files are not found anymore, right-click on the project, Build Path -> Configure Build Path... -> Libraries and remove from the list the missing libraries.
From 1.7 to 1.9 `xml-apis-1.0.b2.jar` was removed.

mat-l, 2017-01-20:
With the new ROSJavaLib it is making progress.
The robot is unlocking when a node is started and we get the following ouput.
[image]
The interesting error message is probably:
The object template "tool" was not found.

Also gazebo and rviz are loaded but the simulated robot is not updated with the current position of the real robot. 
We tried a few different launch files but mostl
y roslaunch iiwa_tool_examples iiwa_tool_command_moveit.launch

SalvoVirga, 2017-01-23:
the iiwa_tool repository was meant to show examples on how to build your own modules, for example, extending the robot description with an attached tool. Therefore, it is not meant to be used as it is.

In fact, that launch file that you are using calls a description of an iiwa 7 with a dummy tool attached on the flange, that is simply called `tool`, and it tries to use it.
As I explain [here](https://github.com/SalvoVirga/iiwa_stack/wiki/attachtool#sunrise), once the tool is in the robot description you have to also created it in Sunrise and its frames accordingly to the naming convention.
Then, in your launch file you can just [say which tool you want to use](https://github.com/SalvoVirga/iiwa_stack/wiki/attachtool#selecting-a-tool) and the right tool will be used in Sunrise as well.

So the error there says it all. That launch file tells Sunrise to use a tool named `tool`, but you have no tool name `tool` created in Sunrise and it tells you exactly that.

To just test if iiwa_stack works for you, run `roslaunch iiwa_moveit moveit_planning_execution.launch sim:=false`. You should then have all the topics to command the robot and read its state.     

If you need to attach a tool to your robot, you can **take inspiration** from iiwa_tool and read the wiki about it, but you will need to create your modules accordingly to your needs. iiwa_tool is not meant to be directly used (at least if you don't want to use a useless pointy tool that sits in my lab).

mat-l, 2017-01-30:
Great, it is working :)
Thank you a lot.

closed 2017-01-30 by mat-l