IFL-CAMP/iiwa_stack #45: Sending joint trajectory to the robot

LinYuChi, 2017-01-10:
Hello,

The ROS interface accepts two kinds of input: joint position and end-effector Cartesian. I wonder if it is possible to send joint trajectory to the robot using this package? or do you have any suggestions on how to implement one? Thank you.

marcoesposito1988, 2017-01-12:
Hi @LinYuChi,

we do not plan to include such a functionality, since it is not offered by Sunrise's SmartServo. Implementing it from scratch (and maintaining it) would be a great effort. 

Is there a reason why you cannot use the standard JointTrajectoryActionController from ros_control? Maybe we could document how to start it without MoveIt.

LinYuChi, 2017-01-12:
Hi @marcoesposito1988 ,

Thank you for the reply. We just got our iiwa 7 arm recently and start to see how to move it through ROS. We will do some motion planning in configuration space to generate a joint trajectory and send them to the robot to execute it. A document, a short guide or a reference on how to do this with iiwa_stack would be greatly helpful.

I am also wondering if it is possible to send joint velocity command to the robot? Thank you.

marcoesposito1988, 2017-01-12:
You can do all of this with MoveIt. Do you have a specific need for which it is not suitable? Otherwise I would advise you to go for it. You can find the related information [here](https://github.com/SalvoVirga/iiwa_stack/wiki/commandrobotmoveit).

marcoesposito1988, 2017-01-12:
Joint velocity command will also be supported soon (we have an implementation we are testing).

LinYuChi, 2017-01-12:
The major reason is that the framework we are using is OpenRAVE, which is kind of overlapping with MoveIt, so I would prefer not to depend on MoveIt. But since MoveIt can do it, I will look into the source code to find out how they do it. I am also looking forward to see the update. Thank you very much.

marcoesposito1988, 2017-01-12:
Sounds legit! I will document how to start the robot with only the JointTrajectory controller, without MoveIt.

LinYuChi, 2017-01-12:
Great! Thank you.

Andreluizfc, 2018-10-26:
> Sounds legit! I will document how to start the robot with only the JointTrajectory controller, without MoveIt.

Hi, Marco. Any news on that? We would like to control the robot via [JointTrajectory](http://docs.ros.org/api/trajectory_msgs/html/msg/JointTrajectory.html) Messages.

If not possible, is there any way to perform this by passing the command direct to the Robot Controller?

Thank you.