IFL-CAMP/iiwa_stack #46: ROS Master won't reconnect after restart

gekoraus, 2017-01-17:
I always start the ROS Application on the Master with `roslaunch`. The first start works fine and it connects to the `ROSSmartServo` application on the Workstation. 
But if I restart the ROS application on the Master, the connection doesn't get established again until I also restart `ROSSmartServo`.

SalvoVirga, 2017-01-18:
Well, that is normal. `ROSSmartServo` creates some ROS(Java) nodes that connect to a `roscore` running on your Linux machine. Launching a ROS application with `roslaunch` automatically starts a roscore (if none is currently running).
So you launch a `roscore`, you start some nodes (it doesn't matter that in this case those are ROSJava nodes and they are on another machine), those nodes connect to the `roscore` and then you kill it. The nodes simply lose the master they are connected to, even if you start a new `roscore` you need to relaunch those nodes as well. That happens with any ROS application if you kill its master.

Solution: keep a roscore always running in one terminal, then you can kill your application but the ROSJava nodes will still have a connection.

I hope I was clear and this can help you!

SalvoVirga, 2017-01-19:
let us know if you have more doubts, if not, you can also close the issue

gekoraus, 2017-01-20:
Thanks for the clarification, I wasn't clear about that. Using your Solution works like a charm!

closed 2017-01-20 by gekoraus