IFL-CAMP/iiwa_stack #47: Sending command/JointTorque to the robot

tdinesh, 2017-01-31:
Currently, Sending command/JointTorque command/JointVelocity to the robot is disabled/commented in iiwaRos.cpp. Is this an untested feature? We have the iiwa 7 and I would like to use this feature. Any gotchas/concerns about enabling these?

ahundt, 2017-01-31:
Sorry for jumping in, would probably need to be done over FRI.

github.com/ahundt/grl also has (untested) torque support and integrates with iiwa_stack. For torque control FRI will most likely be needed.

People have implemented force control there, but didn't fix some items to merge:
https://github.com/ahundt/grl/pull/106

Seems iiwa_stack didn't support it as of last year:
https://github.com/SalvoVirga/iiwa_stack/issues/28

Will you have/need inertial parameters as well? Discussed here:
https://github.com/SalvoVirga/iiwa_stack/issues/14

tdinesh, 2017-01-31:
Thanks @ahundt. This helps a lot

My initial hesitancy for not using grl was due to its build dependency. Does it compile on 16.04?

Could you please elaborate more on "integrates with iiwa_stack". Is this integration for gazebo/moveit packages?

Currently, I have the cabinet KONI disabled for working with iiwa_stack. Looks like I have to revert this or use USB3-Ethernet adapter #7 to keep using iiwa_stack driver. 

Inertial parameters will be definitely needed for the torque control.

ahundt, 2017-01-31:
The number of dependencies has been greatly reduced:
https://github.com/ahundt/grl/blob/master/INSTALL.md

Yes it integrates with moveit + ROS using the iiwa_stack where grl supplies the drivers.
I've mostly used V-REP but will likely be setting up gazebo soon.

useful links for determining intertial parameters of kuka iiwa:

- https://www.google.co.in/webhp?sourceid=chrome-instant&ion=1&espv=2&ie=UTF-8#q=Dynamic+Identification+of+Manipulator%3A+Comparison+between+CAD+and+Actual+Parameters
- http://www.inacomm2015.ammindia.org/img/94.pdf

There is more I'd be happy to disucuss and help with items that are grl specific but that should probably be done by creating an issue under the grl github project.

tdinesh, 2017-01-31:
Thanks! Will give it a try and get back on issues if any.

closed 2017-01-31 by tdinesh

SalvoVirga, 2017-02-02:
Just as comments:
The inertial parameters presented in that paper are now integrated in the URDF of the iiwa7. Still, I cannot guarantee their accuracy.
And yes, we do not provide an interface for joint torque control. We would need to move away from the native ROSJava part on the cabinet and lose quite some features. We still have in mind to have a branch that uses FRI and allows only for joint position and torque control, but since we actually won't need it for our own use it's hard to find the time to work on that. Help from the community that needs that feature would be great :P

That said, I'm quite interested to learn if you manage to use FRI for torque control. Maybe drop me an email if everything works!