IFL-CAMP/iiwa_stack #48: Header file iiwaRos.h not found

broesdecat, 2017-02-01:
When trying to the build the package after cloning, i get an error that iiwaRos.h cannot be found when compiling iiwa_hw.h. Replacing it with the full path gets it to compile.
Probably there is some include missing in the cmake file?

Thx!
Broes

SalvoVirga, 2017-02-02:
well, that sounds quite weird to be honest, I compiled the package on fresh installed 14.04 and 16.04 machines recently with no problems.
Also, the build from Travis compiles everything on a fresh 14.04 right after installing ROS Indigo, and that doesn't fail, so...
Are you using catkin_make or catkin build, right?

SalvoVirga, 2017-02-04:
Still having that problem? Did you try on a fresh installation?

broesdecat, 2017-02-09:
Sorry I could not get back to you earlier.

I retested the issue today and I could not reproduce it, although I did not change my setup (and it already was a fresh installation). As I do not have a clear idea what could have caused it in the first place, I'll just close the ticket.

I will certainly be checking out the new build for the features you added, and I really like the project, thx for the work!

closed 2017-02-09 by broesdecat

davidadley, 2017-02-27:
I have the same issue no idea why, it worked previously

```
Errors     << iiwa_hw:make /root/catkin_ws/logs/iiwa_hw/build.make.000.log
In file included from /root/catkin_ws/src/iiwa_stack/iiwa_hw/src/iiwa_hw.cpp:36:0:
/root/catkin_ws/src/iiwa_stack/iiwa_hw/include/iiwa_hw.h:38:22: fatal error: iiwa_ros.h: No such file or directory
 #include "iiwa_ros.h"
                      ^
compilation terminated.
make[2]: *** [CMakeFiles/iiwa_hw.dir/src/iiwa_hw.cpp.o] Error 1
make[1]: *** [CMakeFiles/iiwa_hw.dir/all] Error 2
make: *** [all] Error 2
cd /root/catkin_ws/build/iiwa_hw; catkin build --get-env iiwa_hw | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

```

Using `catkin build`

davidadley, 2017-02-28:
I think this is likely related to not running `source catkin_ws/devel/setup.bash` correctly.