IFL-CAMP/iiwa_stack #50: Questions about command/JointVelocity

LinYuChi, 2017-02-03:
Hello,

I tried to use the new function introduced in the new update to set the joint velocity by publishing JointVelocity to the topic /iiwa/command/JointVelocity. I am not sure if it is a bug or I have a misunderstanding on what the server will do. This is what I did and saw:

(1) I moved the robot to joint position [0,0.7,0,0,0,0,0] by publishing to /iiwa/command/JointPosition. This works fine.

(2) Then I tried to move the 2nd joint with a velocity -0.2, and other joints are 0 velocity. I expect the second joint to reach -0.2 velocity after a short acceleration and keeps that velocity until new command is received or triggered some safety limits. However, the first time I send this command, it does not move. Everytime I tried this, I need to send exactly 3 times to make it move

(3) After it moves, it seems it will randomly stop at some joint position. I tried several times, and cannot find out where and why it will stop.

(4) After it stops, I tried to rotate the joint back by sending 0.2 velocity in 2nd joint. Usually I will see this error message on the SmartPad and the program stops:

https://1drv.ms/i/s!AvtCHo07178NgfwTaBy0gyovl5Fxxw

I believe I have misunderstandings on what this function does. Could please tell me more details about its implementation? Thank you.

Yu-Chi

SalvoVirga, 2017-02-04:
Hum, it should exactly work how you expected it to work. It was used extensively by a colleague, so it should actually work properly. I will have a look myself asap next week! 

Meanwhile, you can find the actual implementation in [ROSSmartServo](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSSmartServo.java#L472-L495).

SalvoVirga, 2017-02-06:
Indeed something went wrong when fixing another bug, you could try [this branch](/SalvoVirga/iiwa_stack/tree/hotfix/JointVelocity) to check if everything looks alright for you.
I tested it in AUT at full speed, but using an order of magnitude less for the commands, e.g. -0.02/0.02 and it looks good for me.
Let me know and I will merge it immediately to master.

SalvoVirga, 2017-02-09:
Did you have time to test the branch yourself? For me it looks quite good and I would like to merge it as soon as possible. Let me know.

LinYuChi, 2017-02-09:
Hi,

Sorry for the late reply. I just tested the hotfix/JointVelocity branch. It works fine on our iiwa7 now. Thank you.

SalvoVirga, 2017-02-09:
great, thanks a lot for the feedback!

closed 2017-02-09 by SalvoVirga

shrutic12, 2023-03-14:
Hi! I have a confusion regarding the use of  /iiwa/command/JointVelocity command, what is the frame_id supposed to be if we are controlling sya for instance just joint 7? Will it still be iiwa_link_0 or will it be the end effector frame?

exo-core, 2023-03-15:
The header is mostly about the timestamp. It should be fine to put any valid frame name or even leave the field empty.

PR #51 (2017-02-09): Hotfix/joint velocity
Fixes #50