IFL-CAMP/iiwa_stack #53: ROSSmartServo not quitting cleanly - is the reason known?

mbanerj, 2017-02-13:
First of all, iiwa_stack is remarkable, and so is your commitment to continually developing it!

I'm seeing that ROSSmartServo's tendency to leave the Sunrise task running in the background is a known issue. **Has anyone figured out _what_ exactly causes the task to hang in the background?** I had a thought that maybe the SmartServo motion isn't explicitly stopped, so I added _motion.getRuntime().stopMotion()_ in the cleanup() method, but it doesn't seem to help.
 
Beyond the obvious nuisance of waiting 15 seconds for the error message, the memory used to run the application is never relinquished. So if you're doing debugging, starting the application several times leads to an out-of-memory error on the sunrise controller, which requires a reboot to resolve.

mbanerj, 2017-02-14:
Well, I figured out part of it. 

[image]

The thread that fails to quit is something called "XML-RPC Weblistener". Can you recommend a good way to close this thread in the context of ROSBaseApplication?

SalvoVirga, 2017-02-16:
Really good point, we really hate that ROSSmartServo is not cleaned properly and we tried to make sure everything is disposed correctly, but still something is not.

I recently noticed that ROSSmartServo closes properly if a roscore is still running on the Linux side - at least for me on 16.04. Else, there are those 15 seconds to wait and the incredible annoying "PermGen Space" error after too many runs/syncs.

Thanks for points us out to the possible problem! That thread is part of what ROSJava builds, so we might need to have a look in there directly (we provide all the needed jars but one can also use the source code from [the repo](https://github.com/rosjava/rosjava_core) to check what is going on).

We are under deadline, so we won't be able to work on this for the next 2-3 weeks.

mbanerjee1, 2017-02-17:
Thanks for the response. I'll look at the jars' source code and see if I can find the chain leading to XML-RPC Weblistener in the meantime and let you know if I find anything.

mbanerj, 2017-02-17:
Thanks for the response. I'll look at the jars' source code and see if I can find the chain leading to XML-RPC Weblistener in the meantime and let you know if I find anything.

mbanerj, 2017-05-11:
So a quick update, I recently started ramping back up on my iiwa project and was going to follow up on this issue. I see [this update](https://github.com/rosjava/rosjava_core/commit/f4116594ac642536c7a82f7f5a7e32c23add3109) on the rosjava repository that seems to involve some kind of fix to publishers and subscribers on node shutdown. Do you think this would solve the issue we're talking about in this thread?

SalvoVirga, 2017-05-11:
Definitely. I was already following that fix and I have in mind to try the new version since weeks, but still didn't have the time. jars can be found already [here](https://github.com/rosjava/rosjava_mvn_repo), there are also some others that might be worthing upgrading.

mbanerj, 2017-05-16:
One thing you're probably going to run into when you try it - the rosjava source was compiled in a later version of Java (1.8 I think) while Sunrise uses v1.6 for runtime. So the project doesn't want to synchronize if you drop in the .jar files from the repo you linked.

I tried dropping the source files in my ros workspace (like your guide for the messaging) but, after a day of categorically failing to get it to compile properly, have given up. I don't have the expertise to really go any further :( (unless of course you have any suggestions).

SalvoVirga, 2017-05-16:
Oh, I'm sorry you had to fight with this. I actually have a solution, since I had to generate jars in the past. Java 7 should be ok too. The following works for ROS Kinetic and 16.04, which I suggest you to use anyway. 

Install Java 7 (need to add a ppa for 16.04)

```
sudo add-apt-repository ppa:openjdk-r/ppa
sudo apt update
sudo apt install openjdk-7-jdk
```

Switch the default Java to use to Java 7

`sudo update-alternatives --config java`

Install the necessary packages to build rosjava_core

```
sudo apt install ros-kinetic-genjava
sudo apt install ros-kinetic-rosjava-test-msgs
```

Build the workspace where rosjava_core was cloned with `catkin_make`

`catkin build` would not generate the jars.

SalvoVirga, 2017-05-16:
I am quite overwhelmed with others duties until the end of the month, else I would give it a try myself.

mbanerj, 2017-05-16:
No worries, everyone's busy and the thread termination issue isn't your fault anyway.

Your steps are much appreciated! I will try them out tomorrow. I'm assuming that after switching the version to 1.7 for compile, I need to go into my Sunrise project > Settings > Java > Compile and change the compatibility to 1.7 as well? I think that should be the only other thing to change, but I will try it out and let you know if it works.

mbanerj, 2017-05-22:
Unfortunately, this doesn't seem to jibe well with Sunrise. The controller is running Java 1.6, and produces "unsupported major.minor 51.0" as the getLogger output even if you manage to get the project to synchronize properly :(

mbanerj, 2017-06-02:
So I've been messing around with multithreading for my custom additions to iiwa_stack in the last couple of weeks and have learned enough to identify something. This section in ROSBaseApplication.java causes a blocking wait:

```
try {
			getLogger().info("Waiting for ROS Master to connect... ");
---->>>>>	configuration.waitForInitialization();
			getLogger().info("ROS Master is connected!");
		} catch (InterruptedException e1) {
			e1.printStackTrace();
			return;
		}
```
waitForInitialization() theoretically throws an InterruptedException, but I see no way to interrupt the thread programatically. nodeMainExecutor calls .execute, which I believe allows little control over the task itself. I am new to multithreading though, so I fully acknowledge that I may be missing something. 

Anyway, this is *why* it closes properly with roscore running and has an errant task without roscore. If the iiwaConfiguration object can't acquire the , the thread just sits in a blocked state indefinitely while waiting to acquire the semaphore permit from iiwaConfiguration.

SalvoVirga, 2017-07-10:
@mbanerj if you have time, you could check the branch [this branch](https://github.com/SalvoVirga/iiwa_stack/tree/hotfix/CleanDisposal) or 6be0c82. From my experience this completely solves the problem.

commit 454886dd (2017-12-11): Merge pull request #97 from IFL-CAMP/hotfix/CleanDisposal

Fixes #53
closed 2017-12-11 by SalvoVirga