IFL-CAMP/iiwa_stack #54: IllegalStateException when LBRExtTorqueObs high

felixvd, 2017-02-16:
I have a hunch that this is a robot setup issue, but maybe you know where to look to fix it. When ROSSmartServo is running, it exits with an IllegalStateException when the external torque is reported high (see image) 

[image]

From what I can tell we have set up no safety monitoring on the torque, and the code seems like it should not break out. Do you have any experience with this? Should we upgrade the robot's firmware?

We are running Sunrise 1.7 with the code from master-Sunrise1.7

Thanks in advance

SalvoVirga, 2017-02-16:
Are you maybe starting ROSSMartServo with the robot in home position?
This is probably due to [this check in the java code](https://github.com/SalvoVirga/iiwa_stack/blob/master-Sunrise1.7/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSBaseApplication.java#L191-L192), if the robot is in a singularity position it cannot validate the tool/flange for impedance control.
Since anyway ROSSmartServo starts in position control mode, I actually removed those lines and [these other two](https://github.com/SalvoVirga/iiwa_stack/blob/master-Sunrise1.7/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSBaseApplication.java#L191-L192) recently. But I didn't have time to update that branch to the new release, so it is quite behind.
Just start ROSSmartServo from a non singularity position.

I will suggest you in any case to ask KUKA for Sunrise 1.11 + Connectivity, the master branch uses that version and has a lot of new features. I won't have time to bring the branches for 1.7 and 1.9 up to date for the next 2-3 weeks.

Keep me updated!

felixvd, 2017-02-17:
It seems that this was actually due to old tool data being set in the project. If anyone else has this project, set up an empty tool to attach to the flange or just perform the automatic load data determination with nothing on the flange.

closed 2017-02-17 by felixvd

mbanerj, 2017-02-17:
This actually seems to be trickier when you have a tool set as "safety related".
 
If the tool is not set as safety related, it will not be registered automatically when Sunrise starts, so the robot will detect a collision immediately (beyond a certain weight threshold, of course) as soon as it boots up. However, calculateLoadOnFlange() in this case will report 0.0kg after the .attachTo() command is used in my code.

If the tool is set as safety-related, then using .attachTo() in the code _seems_ to attach it "twice", with .calculateLoadOnFlange() reporting a load of [tool weight]. 

KUKA engineering says the proper way is to use .attachTo() regardless of the safety status of the tool. Has anyone else run into this?