IFL-CAMP/iiwa_stack #55: Robot not moving despite network connection working (Sunrise 1.7)

felixvd, 2017-02-17:
We are having trouble moving the robot, although we receive its position and torque data. We double checked the IP and network settings as mentioned in the wiki, and we looked at the network activity in the Sunrise Cabinet. Sent and Received data increases normally when we send a jointPosition, the Sent data stops increasing after the first message is received, it seems.

We tried putting a debug message into the callback function in the Java code, but it is never displayed. The robot interface complains about timeouts in other errors (see pictures), so I wonder if the message arrives but then locks up the thread somehow.

We use [this code](https://gist.github.com/felixvd/18f8a8b9c6f539449037aa37c07be658) to publish the jointPosition. We're new to ROS, so this code might well be the problem.

[image]
[image]
[image]

We are still running on KUKA Sunrise 1.7 and the corresponding branch of iiwa_stack, but we will upgrade soon.

mbanerj, 2017-02-17:
Random question - how were you able to get the SmartHMI to fit on only half of the monitor? Super annoying to have it either be fullscreen or minimized.

felixvd, 2017-02-17:
It's still fullscreen, but it stayed in the background when we alt-tabbed to a different window. I think it only minimizes if you alt-tab to the desktop.

felixvd, 2017-02-21:
We updated to Sunrise 1.11 and version 1.2 of iiwastack, but this problem persists. Is there something in the network settings we can check?

felixvd, 2017-02-21:
To give a small update: We added some debugging lines into the "iiwaSubscriber.java" file as listed below, and they are never run. It would seem that no new messages are received. Anyone have an idea on how to troubleshoot this?

		jointPositionSubscriber.addMessageListener(new MessageListener<iiwa_msgs.JointPosition>() {
			@Override
			public void onNewMessage(iiwa_msgs.JointPosition position){
		        System.out.println("New jp message received");
				synchronized (new_jp) {
					jp = position;
					currentCommandType = CommandType.JOINT_POSITION;
					new_jp = true;
			        System.out.println("Set command type");
				}
			}
		});

felixvd, 2017-02-26:
It turned out to be purely a networking issue. I'm going to quote my colleague on how this was solved:

Problem:
The node in Kuka cannot convert the kukaPCwhateverhostname into 160.69.69.100 and, therefore, cannot get the messages
Best workaround:
1) change the /etc/hostname file contents to 160.69.69.100
This also worked:
2) add an environment variable called ROS_HOSTNAME with the value of 160.69.69.100
export ROS_HOSTNAME=160.69.69.100
roscore
NOTE about (2): for a strange reason (probably a bug in ROS) you cannot just set this variable in the bashrc, cause even when set there, the ros master starts on kukaPCwhateverhostname:11311 instead of 160.69.69.100:11311

closed 2017-02-26 by felixvd

adamconkey, 2020-11-18:
Thank you @felixvd you just saved me a lot of debugging time, I had this exact issue, added option (2) to my bashrc and worked like a charm.

matthias-mayr, 2020-11-18:
I had similar issues when trying to use hostnames rather than IPs and I switched to using IPs only.

The best practice for us is to get one router per robot system that creates a private network for the cabinet, computer, grippers, etc. This way one controls all the IP addresses at one point, the DHCP server on the router and can make that they never change.
That solution also introduces some security, because the router uses NAT and blocks traffic from outside. Remember that the cabinet has an open remote desktop connection with a default password enabled, so everyone in the network can see the content of the SmartPAD and also interact with it.