IFL-CAMP/iiwa_stack #57: Unable to load virtual joints in iiwa14 srdf file

davidadley, 2017-02-22:
Hello,

Running the following command

`roslaunch iiwa_moveit moveit_planning_execution.launch model:=iiwa14`

results in the following warning:

```
[ INFO] [1487789324.400531125, 3.279000000]: Loading robot model 'iiwa14'...
[ WARN] [1487789324.400599896, 3.279000000]: Skipping virtual joint 'fixed_base' because its child frame 'iiwa_link_0' does not match the URDF frame 'world'
[ WARN] [1487789324.400649151, 3.279000000]: Skipping virtual joint 'fixed_ee_kuka' because its child frame 'iiwa_link_ee_kuka' does not match the URDF frame 'world'
[ WARN] [1487789324.400699731, 3.279000000]: Skipping virtual joint 'fixed_ee' because its child frame 'iiwa_link_ee' does not match the URDF frame 'world'
[ INFO] [1487789324.400766146, 3.279000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1487789324.521896939, 3.400000000]: Loading robot model 'iiwa14'...
[ WARN] [1487789324.521933564, 3.400000000]: Skipping virtual joint 'fixed_base' because its child frame 'iiwa_link_0' does not match the URDF frame 'world'
[ WARN] [1487789324.521951315, 3.400000000]: Skipping virtual joint 'fixed_ee_kuka' because its child frame 'iiwa_link_ee_kuka' does not match the URDF frame 'world'
[ WARN] [1487789324.521971997, 3.400000000]: Skipping virtual joint 'fixed_ee' because its child frame 'iiwa_link_ee' does not match the URDF frame 'world'

```
These virtual joints are defined in `iiwa_moveit/config/iiwa14.srdf`and https://github.com/cpaxton/costar_stack uses these joints in its iiwa driver. 

Any idea what the problem might be? There appears to be a mismatch between the world frame and how the links are defined.

SalvoVirga, 2017-02-22:
Can you point me to where/how costar uses those joints? I don't use an iiwa14 since a year or more, so the urdf/srdf might not be up to date with the iiwa7 model.
I'm under deadlines so I might not answer quickly in the next days!

cpaxton, 2017-02-22:
This also might be a costar bug, it's been a while since I've had access to an iiwa14 to make sure everything works properly and we've been aggressively adding features and refactoring. I will say that the simulated version still works with older versions of Salvo's iiwa_stack.

We define base and end links for the kinematic chain of the arm here: https://github.com/cpaxton/costar_stack/blob/master/costar_robot/costar_robot_manager/src/costar_robot/iiwa_driver.py#L41

We use these with TF to set the different waypoints, and publish a transform from base to world in costar_arm.py.

David, does this work without CoSTAR stack? if so it's probably a bug on my end.

ahundt, 2017-02-23:
I might be able to test on the kuka 14 (except no gripper) but it will take a fair amount of effort and couldn't be until next week.

SalvoVirga, 2017-04-03:
@davidadley was this fixed on the costar side?

davidadley, 2017-04-03:
Hi,
I don't think the virtual joints are loaded but costar has been able to work around this.

SalvoVirga, 2017-04-05:
I just removed those virtual joints from the srdf to match the one for the iiwa7. If any other problems should come up regarding this, feel free to reopen the issue.

closed 2017-04-05 by SalvoVirga