IFL-CAMP/iiwa_stack #58: Connection via X66

AsadT, 2017-03-01:
I did not set up the Sunrise cabinet to connect to ROS Master via KONI port. I set the Master as per your instructions. I am using the X66 port The SmartServo App on sunrise connects to the master. I can see the Topics but when I try to echo anyone of them to get the data, I get a "Couldn't find an AF_inet address for [[fe80]" error. 

My questions are. 
1. Is this because I did not setup the cabinet to work with KONI port? 
2. Why are we not using the X66

SalvoVirga, 2017-03-04:
The two questions are linked:

1. Yes
2. We are using the KONI interface rather than the X66 exactly because of the error that you are encountering. What ROSJava uses for its networking is not able to discover the X66 interface, which is somehow natted by KUKA. That is the only reason, I would be nice to just be able to use the X66 but that would require some internal work on ROSJava.

To use the KONI interface there is a small modification to the controller to perform, described [here](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup). It is fairly easy and reversible, it is needed only once and it takes around 5 minutes.

SalvoVirga, 2017-04-03:
I hope the question was answered, if have more doubts please feel free to reopen the issue.

closed 2017-04-03 by SalvoVirga