IFL-CAMP/iiwa_stack #60: Connect IIwa kuka robot with kinect 3d camera

usman2k, 2017-03-03:
I need to connect IIwa kuka robot with kinect 3d camera by using ros.

SalvoVirga, 2017-03-04:
Well, this is not really related to iiwa_stack per se.
If I understood correctly what you mean, I would say that this is a very common problem - calibrate a camera sensor to a manipulator.
Without going into details, I can tell you it is fairly simple and I personally use almost constantly a Kinect calibrated to my iiwa. You can employ any of the hand-eye calibration techniques out there, some are already available for ROS. For example [this](https://github.com/jhu-lcsr/aruco_hand_eye) is a pretty good one, it would definitely do the job and it is based on the classical Tsai-Lenz algorithm. 

I don't know what you background is, but if what I wrote so far doesn't ring some bells please have a look to the literature on hand-eye camera/robot calibration.
I you have more specific questions, write here or drop me an email. Else, this issue could be closed since not of great relevance.

usman2k, 2017-03-06:
Thanks for your kind response....
Actually, I am new in the field of robertics,  I am  doing my graduation these is inwhich in need to do assembly task by using the robot and human collabration with LBR-IIWA kuka my task is bese on Inverse kemamatis while i need to connect the Robos with kematic 3d camera I installed lunix then kematics ros and moveit package but ı am not findlar with ros 
Could you please guide me where i need to start and which computer language ı need to larn on the contıres IIWA Work with Java which ros...

SalvoVirga, 2017-03-15:
It is very difficult to give you a complete answer here :|
If you are starting with ROS, please spend some time learning the basics. There are plenty of tutorials around: http://wiki.ros.org/ROS/Tutorials.
For your LBR iiwa you can use iiwa_stack, for a Kinect v1 have a look at [this](http://wiki.ros.org/freenect_stack) or [here](https://github.com/code-iai/iai_kinect2) for a Kinect v2, not sure which one is the best driver for the v2 tho.

If you need to perform a camera-robot calibration, I pointed you to a good solution in my previous post.
I hope my indications can help you, unfortunately I just cannot find the time to guide you through all the single steps.

SalvoVirga, 2017-04-03:
I hope my comments were helpful, good luck for your project. Feel free to ask more here or privately if you need help with iiwa_stack.

closed 2017-04-03 by SalvoVirga

usman2k, 2017-04-13:
Thanks for your kind response.... I learnt ros basic tutorial.... Actually I need to make some assembly like below mentioned link
https://youtu.be/tCss_TjSs7Y
Someone told me to use ros moveit packages because its inverse kinematics process. I want to know your suggestion.

Best Regards,
Usman Ishtiaq