IFL-CAMP/iiwa_stack #62: Examples of iiwa_stack Sunrise files extended with custom functionality?

mbanerj, 2017-03-08:
It seems that I need to use your ROSSmartServo application as a base, and add my own publishers, subscribers, messages, and control loop elements. Are there any examples of implementations of iiwa_stack in this manner? My goal is to be able to make my modifications modular, so I can get new versions of iiwa_stack easily, but I'm curious to know if there are several possible ways of doing this.

SalvoVirga, 2017-03-15:
Hum, if what you need is something not too specific to your use case and if you can share it, the best scenario would be that your additions would be merged into this repo when they are done. 
But if that is not the case, I would rather create distinct classes to handle your needs. Have a look on how [iiwaSubscriber](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/iiwaSubscriber.java) works and how it is added to ROSSmartServo, for example. You can easily create additional subscribers and publishers in the same way and have only few different lines in the main application.
You can also make your own robotic application instead of modifying ROSSmartServo, just derive one from [ROSBaseApplication](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSBaseApplication.java).
Same for new messages, you can create your own xxxx_msgs package and add those definitions instead of modify iiwa_msgs. You would need to generate the Java code for that, in Indigo you can do it [like this](https://github.com/SalvoVirga/iiwa_stack/wiki/addmessage), in Kinetic is even easier.

But again, ideally your additions could just be part of iiwa_stack :)

mbanerj, 2017-03-15:
Thanks! So I figured out how to do it, and it's unfortunately pretty specific to my case, but it turns out I did exactly what you said. I've got xxx_Config, xxx_Publisher, xxx_Subscriber, etc that all extend AbstractNodeMain and are, and a xxx_main class that is basically a duplicate of ROSSmartServo, but with my additions that run first at the beginning of controlLoop(). I don't _entirely_ like the idea of maintaining a duplicate of ROSSmartServo components, but I figured that manually maintaining changes to one file is at least better than trying to do the whole thing.
 
One very low hanging fruit I think would help to add to iiwa_stack from a general standpoint is adding a placeholder method in ROSBaseApplication immediately before controlLoop() that can be defined as needed by a subclass, maybe something like addOnLoop(). Ideally this would have its own decimation counter and controlDecimation, but that may be easier once those values are moved over to config.txt per the source comments. I'm just starting out using git so I don't want to mess anything up... if you can tell me the best way to go about adding this to the main iiwa_stack repo I'd be happy to give it a shot.

SalvoVirga, 2017-03-20:
You could definitely work on your fork of iiwa_stack (on a specific branch) and point me at your additions or open a pull request once your tested everything. No worries, there is no way you can mess things up :)