IFL-CAMP/iiwa_stack #63: Does the package support multiple robot arms?

broesdecat, 2017-03-17:
Hi,

I'm still in the learning-ROS phase, so please correct me if I am going about this in the wrong way.

I was trying to get two iiwa robotarms spawn in gazebo so I can control them separately.
From the way the package is (nicely) structured, I would have the feeling that it should be possible with minimal changes, but I cannot get it to actually work.
From what I find online, the best guess seems to be to adapt the urdf to get multiple arms into a single file, so that's what I tried.
What I did:
- adapt the transmission.xacro so all names are robot_name dependent
- spawn 2 iiwa's in the iiwa14.urdf.xacro, with a different robotname
=> this works, and both arms get spawned
To try to control them, I 
- spawn the iiwa_control twice, with a different robotname, and the robot_name is also part of the controller names.
- adapted the yaml file so it has a variant of each controller for each robot_name
With that, I can almost control each arm separately, except that 2 joint_states are published, which each contain all joints, so they constantly contradict eachother and the arms jump between positions all the time.

I hope this explanation makes more or less sense :)

Thanks for any help!
Broes

SalvoVirga, 2017-03-20:
Hum, I never tried myself what you wrote sounds correct.

> 2 joint_states are published, which each contain all joints, so they constantly contradict eachother and the arms jump between positions all the time.

Are the two joint_states publishes in two different topics? (They should)
Also, the two robots should have different names for their joints, as the names depend on the robot_name, is that the case?

broesdecat, 2017-03-21:
I committed my changes in a fork of your repo, so maybe you can immediately see from the changes what might be going wrong (very few changes): https://github.com/broesdecat/iiwa_stack

The jointnames are indeed different, and the joint_states are published in different topics, but they both contain all 14 joints.

With the changes in my fork, if you for example publish a command to /iiwa/PositionJointInterface_J2_controller, the rviz visualisation will start jumping between 2 positions, and gazebo will not do anything. If you publish to the other robot (/iiwa2/PositionJointInterface_J2_controller) nothing happens.

SalvoVirga, 2017-04-05:
I had some time to have a look at your changes.
In general, you should try not to directly change the packages in iiwa_stack, but rather use them in your own packages for your needs. With #68 I made the origin of the robot model parametric, so that one can change were the robot is spawned.

I made very quickly an example of how to spawn two iiwas in Gazebo [here](https://github.com/SalvoVirga/iiwa_stack_examples/tree/master/two_iiwa). 
There are mainly two launch files to look at:
- [here](https://github.com/SalvoVirga/iiwa_stack_examples/blob/master/two_iiwa/two_iiwa_gazebo/launch/two_iiwa_world.launch) the Gazebo world is created and two iiwas are spawned with different origins and different name (and in a different namespace).
- [here](https://github.com/SalvoVirga/iiwa_stack_examples/blob/master/two_iiwa/two_iiwa_gazebo/launch/two_iiwa_gazebo.launch) the first launch file is called and the trajectory controllers are created for the two iiwas. Each one is created in the proper namespace, so that they look for the right robot model.

To create the right controller, [this](https://github.com/SalvoVirga/iiwa_stack_examples/blob/master/two_iiwa/two_iiwa_control/launch/two_iiwa_control.launch) launch file is called. This is actually exactly the same as the one in iiwa_control, what changes is that in that package there are two yaml files for the two robots. Sadly, you have to create a yaml file that has the right joint names accordingly to the ones created by the Xacro with the given robot_name, but it'is just a matter of copy/paste.

There are definitely other ways to achieve the same thing, this is just a quick example but it should give you the main idea. If you need to use MoveIt!, then you should probably generate another Xacro that contains both models and make a MoveIt! package out of it, a good example is [this repo](https://github.com/CentroEPiaggio/vito-robot) from the guys in Pisa. They have a robot model with two LBR 4+, but it kinda the same.

I hope this can help, let me know if you need anything else or if the issues can be closed!

broesdecat, 2017-04-07:
Hi Salvo, 

I had a look at your changes and the new example, and that was indeed exactly what I needed to get it working! Screenshot of the result of adding a launch/yaml file generator on top of the whole:
[image]
Still have to figure out why the textures have run away though.

To get it working with moveit i'll indeed have to combine some stuff into one xacro, but that will be no problem, it was more the controllers that I did not manage to get right.

Feel free to close the issue, thx!
Broes

PS: setting_control_mode no longer compiles with me, apparently because something changed in the ConfigureSmartServo submessages?

broesdecat, 2017-04-07:
Found texture issue: in iiwa_description/urdf/iiwa.gazebo.xacro, iiwa_link should be replaced ${robot_name}_link everywhere.

Broes

SalvoVirga, 2017-04-09:
Thanks for the hint, I fixed that. I close the issue, I will link it in the wiki as a reference for other people that want to use multiple arms.

> PS: setting_control_mode no longer compiles with me, apparently because something changed in the ConfigureSmartServo submessages?

It is not clear what you mean with this. Does something in iiwa_ros not compile for you? Did you try to clear the workspace? Nothing change in the message definitions since some months.

closed 2017-04-09 by SalvoVirga