IFL-CAMP/iiwa_stack #64: iiwa7 xacro file error

LinYuChi, 2017-03-23:
Hi,

There are some transform error in the iiwa7 xacro file cause link 3 of the robot not rotating to the joint 3 center. I saw this small displacement when I rotates joint 3:

[image]

Later I found that there are some overlapping in link 1 and link 2:

[image]

By changing the origin of link 2 from (0,0,-0.006) to (0,0,0) and link 3 from (0,-0.005,-0.026) to (0,0,-0.026). The problem goes away without changing the transform of the links.

I tested both in OpenRAVE using or_urdf, and in RViz, they showed the same result.

SalvoVirga, 2017-03-27:
Great! I was aware of this visual error, but since the transforms were still correct I didn't want to play around with the meshes. I would be the best if you could open a pull request with the changes, else I will do that myself in the next days.

SalvoVirga, 2017-04-03:
fixed with #66

closed 2017-04-03 by SalvoVirga