IFL-CAMP/iiwa_stack #67: JOINT_VELOCITY, only sets position?

jenskober, 2017-03-31:
Hi,

I just took a look at the implementation of JOINT_VELOCITY in ROSSmartServo.java but haven't actually tested it yet.
In contrast to JOINT_POSITION_VELOCITY in setDestination() only the position is set, but not the velocity. 
Did that work better in practice? I think otherwise you could just do

`helper.rosJointQuantityToKuka(commandVelocity.getVelocity(), jv);`
`for(int i = 0; i < robot.getJointCount(); ++i) jp.set(i, jp.get(i) + loopPeriod*jv.get(i)); //add the displacement to the joint destination.`
`...`
`motion.getRuntime().setDestination(jp, jv);`

similar to JOINT_POSITION_VELOCITY

Thanks,
Jens

SalvoVirga, 2017-04-06:
This is a good point, I would need to do some tests myself. I you happen to check that by yourself let me know.

SalvoVirga, 2017-06-30:
I checked that today and it looks like adding also the destination velocity makes the motion a bit smoother (at least empirically), it definitely doesn't harm to add it. Probably there on the next "release".