IFL-CAMP/iiwa_stack #71: Joint Control in gazebo

jocacace, 2017-04-08:
Hi to all,

I installed iiwa_stack on my real robot kuka iiwa and it works perfectly. I am trying to have a control mapping with the gazebo simulator. I am able to start the simulation and use it with moveit packages released in this repository. Sadly I am not able to have the same topic to control single Joints (like the real robot). Is possible to control robot joints directly?

What I do is only to launch the iiwa_gazebo.launch. Should I run other pakages insted? 

Regards.

SalvoVirga, 2017-04-10:
That launch file in iiwa_gazebo also spawns the controllers defined in iiwa_control. By default you will have a JointTrajectoryController, to which you need to [send a JointTrajectory](http://wiki.ros.org/joint_trajectory_controller#Sending_trajectories). 
If you use the argument 'trajectory:=false' you will instead have 7 position_controllers/JointPositionController, one of each joint of the robot. You can then send commands for the single joints, but you will have 7 topics to publish to.
I think there were also attempts around to make a single controller that handles n controllers,  but I guess the easiest way is to use the JointTrajectoryController. With MoveIt! you can just set the values for the joints and send the generated plan.

jocacace, 2017-04-10:
Hi SalvoVirga. Congratulation for your work and thank you for your reply! It's working now. I want to control single Joints only to implement an on-line kinematic controller, anyway Moveit! works great! 

Cheers.

SalvoVirga, 2017-04-10:
Glad it works!

closed 2017-04-10 by SalvoVirga