IFL-CAMP/iiwa_stack #74: How to connect LBR IIWA Kuka with ROS by using IIWA-Stack

usman2k, 2017-04-30:
Could you please describe me step by step process connect LBR IIWA Kuka(7R800) with ROS (Indigo) by using IIWA-Stack.. And it will be more fruitful if there will be some video tutorial regarding this...

SalvoVirga, 2017-04-30:
Did you have a look at the [Wiki](https://github.com/SalvoVirga/iiwa_stack/wiki) already?
There I tried to describe all the setup process.

usman2k, 2017-04-30:
Could you please send me the link

SalvoVirga, 2017-04-30:
I just did in the previous comment... -> https://github.com/SalvoVirga/iiwa_stack/wiki
It is also in the readme of this repo

usman2k, 2017-04-30:
Thanks alot for your kindness...

Is there any video tutorial regarding this... It will be more convenient to understand the process...

Best Regards,

Usman Ishtiaq

[https://avatars1.githubusercontent.com/u/2053024?v=3&s=400]<https://github.com/SalvoVirga/iiwa_stack/wiki>

Home · SalvoVirga/iiwa_stack Wiki · GitHub<https://github.com/SalvoVirga/iiwa_stack/wiki>
github.com
HOME. This wiki aims to give all the information needed to use and extend the ROS packages for the KUKA IIWA LWR. It is aimed to readers familiar with both ROS and ...

SalvoVirga, 2017-04-30:
No videos at the moment, the instructions on the wiki should be clear tho.
Open an issue on specific problems/doubts if you have any during the setup.

closed 2017-04-30 by SalvoVirga