IFL-CAMP/iiwa_stack #75: rostopic echo for readrobotstate doesn't display anything

saunair, 2017-06-23:
Hi, 
I am using my robot with sunrise 1.11 and the standard ethernet interface v(Not KONI). I am able to ping also see the list of topics. But no messages seem to be published to the topics. I have roscore running and rosservo running on the controller. The last message shown on the dialogue box is "entering node loop" after "connected to master"

SalvoVirga, 2017-06-24:
I'm actually surprised that the ROS master gets connected using that ethernet interface.
Could you report here the networks settings on the Linux machine, within Sunrise and the content of the config.txt file?

saunair, 2017-06-25:
I'll share the config file by Monday. 
The setting is: A switch that connects to the controller, Sunrise workstation and the ROS master PC. Every PC is able to ping itself. I haven't changed a lot from the original config file. I've just changed the IP address of the ROS master. 
Also, what I thought is I can use the Ethernet interface(With FRI working on it). The current open issues/enhancements made me assume that I can choose not using the KONI interface. If that's wrong, I'll see if I can get the KONI interface up and working.

saunair, 2017-06-26:
In my ROS-master's bash:
sudo ifconfig eth0:1 172.31.1.146 netmask 255.255.255.0 up
export ROS_IP=172.31.1.146
export ROS_MASTER_URI=http://$ROS_IP:11311

IP of the controller: 172.31.1.150
IP of the workstation: 172.31.1.145

Master and controller are connected through ping (checked by pinging the controller. Could only check one sided)

config file:
robot_name: iiwa
master_ip: 172.31.1.146
master_port: 11311
ntp_with_host: false

mbanerj, 2017-06-28:
Shot in the dark - did you follow the cabinet_setup instructions that reassign the KONI port from the RTOS interface to Windows?

SalvoVirga, 2017-06-28:
From what I got, he is trying to not use the KONI port, but the one used for Sunrise (I think it's called KLI).

When you say:
> IP of the controller: 172.31.1.150
> IP of the workstation: 172.31.1.145

1. I guess the IP of the workstation is 172.31.1.146, since that is the output of ifconfig, I guess it is just a typo.
2. As IP of the controller do you mean the one of the KLI port? i.e. the IP you use in Sunrise to sync thr project

SalvoVirga, 2017-06-28:
Another question to @saunair , when you start ROSSmartServo you should get printed out the IP to which you are connecting. What do you get? A photo or screenshot is even better.

saunair, 2017-06-28:
Hi, 
Sorry for being unclear in my previous comment. 
The workstation means the sunrise PC and the ROS master is the machine I am working with. 
The workstation IP is 172,31,1,145; 
ROS master 172.31.1.146; 
Controller(KLI): 172.31.1.150
The screen-shot as you asked:
[image]

saunair, 2017-06-29:
Do you think I should shift to the KONI interface? Why I didn't do it in the first place is that, I just don't want to add new connections and break the currently working FRI on the KLI(Another project depends on it). Will adding the new KONI interface affect the other connections? 

Question: I won't be able add a new network with the KONI. The new IP will be in the same subnet and IP. I hope that will work.

SalvoVirga, 2017-06-30:
Thanks for the picture, now everything is clear.
The answer is in the value for "Robot IP:" in the picture, which in your case is some kind of IPv6 maybe. That is because the KONI interface still belongs to the RTOS and it is not accessible from Windows/Sunrise and the only interface found is the virtual one between Win and the RTOS (which is used internally).

So yes, you should switch to the KONI interface following [the setup step in the Wiki](https://github.com/SalvoVirga/iiwa_stack/wiki/cabinet_setup). Plug a monitor, keyboard, and mouse to the cabinet and it will take you 5 minutes once. **Be sure to not have a project installed on the cabinet that uses FRI first. If you have one, first install on the cabinet a dummy Sunrise Project with nothing on it.**

You have another project using FRI? If so, what do you use FRI for? I suppose reading the robot joint position and/or torque only, maybe controlling the robot via joint commands (if so, let me know how that works :D). In any case, using iiwa_stack you have also those functionalities (and more), you should be able to modify the working projects in very little time. Just use the APIs in iiwa_ros and you probably reduce the code you need to write to a minimum.

> Question: I won't be able to add a new network with the KONI. The new IP will be in the same subnet and IP. I hope that will work.
Never tried but it should be totally ok, I just prefer to have a dedicated connection, additional network cards are super cheap.

Let me know how it works

saunair, 2017-07-07:
I get the messages now! Followed the instructions as given and replaced KLI with the KONI as the ROSMaster interface. Thanks a lot for following up!

Best of luck with adding the KLI as the interface for ROS in the future.

SalvoVirga, 2017-07-08:
Glad that it worked!

closed 2017-07-08 by SalvoVirga