IFL-CAMP/iiwa_stack #76: Error while trying to run ROSMonitor

saunair, 2017-06-23:
The node gets connected to Ros master, later after I try executing the node further I see this error.
[image]

SalvoVirga, 2017-06-24:
I see, in d2e719db824eecebda2ccbd541372a08faeb47af I removed calls to `SmartServo.validateForImpedanceMode()` from [ROSBaseApplication](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSBaseApplication.java), since that is necessary only when using impedance control and the robot starts in position control.
In ROSMonitor the robot is immediately set to joint impedance control, so that call is necessary there. A fix should be adding it in line 90, I will try to do that on monday.
Thanks for noticing!

saunair, 2017-06-25:
Great! Thanks for the quick reply.

SalvoVirga, 2019-06-05:
ROSMonitor doesn't exist anymore after #178 !

closed 2019-06-05 by SalvoVirga