IFL-CAMP/iiwa_stack #77: Support for SmartServoLIN

SalvoVirga, 2017-06-30:
KUKA provides APIs for enforcing the motion of the robot to be linear (SmartServoLIN), I believe from Sunrise1.11 on.
For some reasons, this type of motion is quite different from the classic SmartServo. e.g.: It doesn't allow setting the relative joint speed at runtime and I have to check how it work with Impedance mode. 
This will make a bit messy the switch at runtime between classic SmartServo and SmartServoLIN.
Currently working on it on [this branch](https://github.com/SalvoVirga/iiwa_stack/tree/feature/SmartServoLIN)

felixvd, 2017-07-01:
I tested this with our robot and while there were some crashes we could not reproduce, it seemed to be working fine.

felixvd, 2017-07-06:
I am testing [this commit](https://github.com/SalvoVirga/iiwa_stack/commit/b9eb9429efda37d07643f1f5e532ed79ffa10169) and have some issues:

- When I let go of the enabling switch in ROSSmartServo while a linear motion is in progress, this error occurs:
[image]

- Sometimes after starting up ROSSmartServo, the first motion causes a fairly violent emergency stop with the same error as above.

- After the pathParametersLin service is called, the timeToDestinationService no longer works and returns -999 until the motion in the cabinet is regenerated (e.g. by using a PTP motion and then a LIN motion again). It can still move, but the speed is not affected. I tested sending the linear values 0.05, 5 and 50 in all axes.

- This is not a bug, but from a user standpoint, it is unfortunate that the path parameters of a motion type can only be set when it is active. This makes it so that the first LIN motion after a PTP motion can not be parametrized before it is requested.

felixvd, 2017-07-06:
Regarding the second-to-last error, I wonder about line 186 in controlModeHandler:

`linearMotion = (SmartServoLIN) controlModeHandler.switchSmartServoMotion(linearMotion, request);`

It seems like the function returns a SmartServo object that is cast to SmartServoLIN. Can that work?

SalvoVirga, 2017-07-13:
Thank you for the report, point by point:

- I doubt this has anything to do with iiwa_stack, I will ask KUKA about it. As usual, their errors are cryptic.
- Never happened to me, if you can manage to replicate it let me know.
- True! I just fixed that in 34849c0
- In 34849c0 I changed the services such that they can be called even if the motion is not currently use, thanks for the suggestion

SalvoVirga, 2019-06-05:
In #178

closed 2019-06-05 by SalvoVirga