IFL-CAMP/iiwa_stack #78: Support for CartesianVelocity command

SalvoVirga, 2017-06-30:
as title

bogdan-paulis, 2017-08-31:
Does this mean that you are working on Cartesian Velocity control, or just changing the speed for position control?
I am interested if there any way to change the default Cartesian Velocity for Position control, during runtime.

SalvoVirga, 2017-09-25:
Hi, sorry for the very late reply! I'm out of office atm.
This specific issue is to add Cartesian Velocity control.

You should be able to change the Cartesian velocity in "classic" position control using the [configureSmartServo](https://github.com/SalvoVirga/iiwa_stack/wiki/configuresmartservo) service. KUKA provides a very strange interface for that, when calling that service you should add [CartesianControlModeLimits](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_msgs/msg/CartesianControlModeLimits.msg), there is a **max_cartesian_velocity** field. If I remember correctly, it would only work when setting Cartesian Impedance mode, if you need precise position control just set the stiffness values to the maximum.

Please feel free to ask / comment more or close the issue if that works.

bogdan-paulis, 2017-09-26:
Thank you for the reply. I do not really see the difference between setting different values for the max velocity and if the max limit is too low, instead of the robot slowing down, it just gives me an error that the speed limit was exceeded.

lyh458, 2021-06-15:
Hi, thank you for you awesome works!
I am now using the latest version of this package, when I connect to iiwa, I can get the ``/iiwa/command/CartesianVelocity`` topic, and I try to publish some message to it, e.g.
```
rostopic pub -r 100 /iiwa/command/CartesianVelocity geometry_msgs/TwistStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: 'iiwa_link_0'
twist:
  linear:
    x: 0.1
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0" 
```
but it seems that the robot did not move with what I desired and very slow. 
Then I try to increase the value to 10, speed seems increased but not smooth
```
rostopic pub -r 100 /iiwa/command/CartesianVelocity geometry_msgs/TwistStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: 'iiwa_link_0'
twist:
  linear:
    x: 10.0
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0" 
```
Are there any wrong parameters setting in my demo? What's the default unit of linear (mm? m?) and angular (radian/s?)? And what is the best frequency that this topic publish at?
If possible, could you please provide any updates of the [documentation](https://github.com/IFL-CAMP/iiwa_stack/wiki/commandrobot)?

Thanks!

danivelikova, 2022-12-29:
> Hi, thank you for you awesome works! I am now using the latest version of this package, when I connect to iiwa, I can get the `/iiwa/command/CartesianVelocity` topic, and I try to publish some message to it, e.g.
> 
> ```
> rostopic pub -r 100 /iiwa/command/CartesianVelocity geometry_msgs/TwistStamped "header:
>   seq: 0
>   stamp:
>     secs: 0
>     nsecs: 0
>   frame_id: 'iiwa_link_0'
> twist:
>   linear:
>     x: 0.1
>     y: 0.0
>     z: 0.0
>   angular:
>     x: 0.0
>     y: 0.0
>     z: 0.0" 
> ```
> 
> but it seems that the robot did not move with what I desired and very slow. Then I try to increase the value to 10, speed seems increased but not smooth
> 
> ```
> rostopic pub -r 100 /iiwa/command/CartesianVelocity geometry_msgs/TwistStamped "header:
>   seq: 0
>   stamp:
>     secs: 0
>     nsecs: 0
>   frame_id: 'iiwa_link_0'
> twist:
>   linear:
>     x: 10.0
>     y: 0.0
>     z: 0.0
>   angular:
>     x: 0.0
>     y: 0.0
>     z: 0.0" 
> ```
> 
> Are there any wrong parameters setting in my demo? What's the default unit of linear (mm/s? m/s?) and angular (radian/s?)? And what is the best frequency that this topic publish at? If possible, could you please provide any updates of the [documentation](https://github.com/IFL-CAMP/iiwa_stack/wiki/commandrobot)?
> 
> Thanks!

Hey, I'm currently facing the same issue as you have before. 
Did you end up figuring out what the problem was?
Thank you in advance for any help!

MShields1986, 2023-08-29:
I've done several tests on this for the linear velocity and it appears to be mm/s, but would let the authors confirm.