IFL-CAMP/iiwa_stack #79: LED control service

felixvd, 2017-06-30:
The Media Flange Electric model of the iiwa comes with an LED that can be controlled like this: 
```
private Controller controller;
private MediaFlangeIOGroup LEDgroup;
 
LEDgroup = new MediaFlangeIOGroup(controller);
LEDgroup.setLEDBlue(true);
ThreadUtil.milliSleep(2000);
LEDgroup.setLEDBlue(false);
```

A service to access it through iiwa_stack would be very fancy :)

The MediaFlangeIOGroup It might only be available if the project is generated for a Media Flange, but then you should have access to the class for “MediaFlangeIOGroup” in the “com.kuka.generated.ioAccess” package.

SalvoVirga, 2017-06-30:
I will check what I get myself in Sunrise, since I don't have a touch Flange. Maybe I could access a robot with that tho.

felixvd, 2017-07-06:
Apparently there are three booleans that can be set separately in the MediaFlangeIOGroup (setLEDRed, setLEDGreen, setLEDBlue) for a total of 7 colors. My auto-generated file doesn't show all of those at the moment though.