IFL-CAMP/iiwa_stack #87: Deployment of sunrise project failed

aravindsiv, 2017-10-04:
Hi,

I followed all the instructions on the [wiki](https://github.com/SalvoVirga/iiwa_stack/wiki/sunrise_project_setup) to set up the robotic applications on our KUKA arm, and after installing, and synchronizing the project, I received the following error:

"A communication error occurred during synchronization with the controller with the IP xxx.xx.xx.xx:8090. Please ensure the controller is reachable via network."

I'm not exactly sure what could have gone wrong here, and I appreciate any of the authors trying to help me out. Looking forward to using your awesome software!

aravindsiv, 2017-10-05:
For reference, I'm on Sunrise Workbench 1.11.1.10

aravindsiv, 2017-10-05:
Follow-up:

The network issue seems to have been fixed (assigning a static IP to the robot seems to have done the trick). Now, I'm facing a new issue, synchronizing the project with the KRC fails because there is a compilation error in the project.

One of the errors that seems to be prevalent is that the package "com.kuka.connectivity" isn't found. Is it possible that the name of the connectivity module has changed in 1.11.1.10?

SalvoVirga, 2017-10-05:
Did you install the Sunrise Connectivity (Servoing) module? It usually comes as an extra zip file that needs to be installed via Sunrise Workbench.

aravindsiv, 2017-10-05:
Oh, I didn't realize that it came as an extra zip file. Is it the .kop file present inside the "WorkVisualAddOn folder?

SalvoVirga, 2017-10-06:
No, it should be on its own and it should be clear where to find it. I believe you get an individual CD from KUKA at the beginning. Connectivity is an extra package that customers usually pay for, you should contact your KUKA representative if you didn't get it.

aravindsiv, 2017-10-13:
We just received the Connectivity package from KUKA, and though the individual java files in src/de do not seem to have any errors, the project itself has some compilation error which prevents synchronization with the KRC. 

Any ideas what could be going wrong? 

Once again, thank you for your time!

aravindsiv, 2017-10-13:
Was able to fix the above issue -- I downloaded the package directly from GitHub...and some Java source file pointers were downloaded by mistake (e.g. ".__ROSSmartServo.jar"), and having not worked with Java before, took me some time to figure out. :|

Now, my smartHMI has an open dialog, "Invalid IP address of smartPAD". The dialog box itself says "The smartPAD has been locked against user interaction. The controller cannot access the smartPAD by the provided IP address". This stops me from synchronizing the project. 

Once again, thanks for putting up with this issue -- once I'm able to synchronize and run the sample application, I'll provide an update and close it.

aravindsiv, 2017-10-13:
After successful synchronization of the project, after restarting the controller, the smartPAD's HMI does not recognize the robot when the keyswitch is turned to the left. When it is turned to the right, though, the robot seems to be detected, and I am given the option to choose between T1, T2 and AUT modes. 

Any idea what could be happening?

aravindsiv, 2017-10-14:
This is the error message displayed by the command line running C:\KRC\ApplicationServer\AppServer.exe:

[image]

aravindsiv, 2017-10-15:
For some reason, re-synchronizing the project and restarting the controller seemed to fix this.

closed 2017-10-15 by aravindsiv

Dylan-CDL, 2023-03-22:
Hi there, 

I think I got a similar issue as the one you mentioned in the beginning.

We were restoring the image files on our KUKA robot. While we attempted to install packages and configure the robot in order to let it communicates in ROS. The communication issue appears to be between my Sunrise Workbench computer and the KUKA robot via port X66. By default, the KUKA robot has its own static IP address (172.31.1.147) with a subnet mask of 255.255.0.0, and my Sunrise Workbench laptop has a static IP address set to 172.31.1.150 with the subnet mask: 255.255.0.0. These two devices seem to be following communication rules, but please feel free to correct me if I have missed anything. 

We followed that Wiki to configure the KUKA. We were able to successfully ping the robot and install the packages in Workbench (1.13) the first time, but after the robot rebooted automatically, it could no longer be pinged or connect with Workbench. We tried to use a second computer with IP 172.31.1.150 and subnet mask: 255.255.0.0 to ping the robot, no response. Additionally, we also tried restoring the KUKA image files again and using that second computer with Sunrise Workbench and repeated the process. It could ping the robot initially and install packages successfully the first time, but after that, it disconnected and there were no responses when attempting to ping or synchronize. This is very strange...

If you have any ideas on what could be wrong, I would really appreciate that! 

> For some reason, re-synchronizing the project and restarting the controller seemed to fix this.