IFL-CAMP/iiwa_stack #88: Connectivity split into FRI and Servoing in Sunrise 1.13

miguelprada, 2017-10-21:
I've recently asked for a quote for iiwa robots and apparently the Sunrise.Connectivity (listed as a requirement in the wiki) module has been split into Sunrise.FRI and Sunrise.Servoing. The quote we got listed these modules as versions 1.13.

Has anyone used these packages with Sunrise version 1.13? Any idea whether they are compatible? And if they are, which of the modules is required: FRI, Servoing, both?

SalvoVirga, 2017-10-22:
We are still using Sunrise 1.11, but 1.13 is out since a while. I will ask for the updated version in the near future, but I expect iiwa_stack to work already there or with little changes. No worries.

Sunrise.FRI and Sunrise.Servoing are indeed two different things, currently iiwa_stack only uses Servoing but there are plans to also integrated FRI (partial discussion on that in #7 ) - someone already uses it. By the end of the year I would like to have that done.

At the end of the day it really depends on what you need, if you need a 1kHz command rate then you need FRI and you will lose a bunch of hand features implemented already in Servoing.

If the price difference is not that high, I would take both, just to not limit yourself in the future.

miguelprada, 2017-10-23:
Thanks, that was very helpful!

closed 2017-10-23 by miguelprada