IFL-CAMP/iiwa_stack #89: Best way to impose external forces

alessmosca, 2017-10-31:
Hello , 
for my research I would like to reflect forces measured by a remote sensor, on the kuka iiwa. 
What is in your opinion the best way for achieving my goal ? 

Example: 
the sensor is measuring a force along 5 N. the iiwa will perceive that force and the effect will be a movement along that axis

SalvoVirga, 2017-11-10:
Hi, it is not totally clear to me how the robot is intended to react to this measured force information.
You have some force sensor (a haptic device?), it feels a force of some N along one axis. Do you want the robot to apply the same force along the same axis? Does it have to move - a fixed length movement? of a length that is relative to the sensed force? continuously in that direction, while the force is there? Or else?

In general, iiwa_stack supports most of the functionalities of KUKA's SmartServo, so Joint/Cartesian Impedance or DesiredForce control might be of use for you. But depending on what you want to achieve, you might want to have a look at torque control, that would be achievable using FRI which is not yet (hopefully not for so long once I have the time to work on that) supported by iiwa_stack.

SalvoVirga, 2017-12-19:
any updates? can this be closed?

alessmosca, 2017-12-20:
sure , no news !
​
A​

closed 2018-02-12 by SalvoVirga