IFL-CAMP/iiwa_stack #90: Could not create controller 'KUKA_Sunrise_Cabinet_1'

aravindsiv, 2017-11-03:
Hi,

I'm facing an issue similar to the one described in the middle of the comments section of #13 where I get the following error on C:\KRC\ApplicationServer\AppServer.exe:

[image]

As a result, on my smartPAD, I cannot see any connected robots, as shown below:

[image]

Any idea what could be going on, and what I can do to fix it? When I encountered this earlier, a simple reboot of the controller helped, though that no longer seems to be the case.

Thanks in advance!

aravindsiv, 2017-11-03:
Found a fix for this (albeit temporarily):

Just connect a monitor, keyboard and mouse to the controller, and close the C:\KRC\ApplicationServer\AppServer.exe window. Once you close it, follow the same path and restart the AppServer (don't use the desktop icon, it doesn't work for some reason).

Waiting to hear back from you guys so that I can close this issue for good.

SalvoVirga, 2017-11-10:
It sounds like a random behavior from the KUKA controller, if you are ok with it you can close the issue. Still, it might be a good reference for someone who encounter the same problem.

closed 2017-11-10 by aravindsiv