IFL-CAMP/iiwa_stack #91: Without SmartServo Motion packages

yajunxuejue, 2017-11-06:
Hi, when I load a Sunrise project from the controller, and then open the .cat file by Sunrise Workbench. There are no "Direct Servo Motion Extension", "Smart Servo Motion Extension" in my .cat file. Therefore, I need to install those packages by Sunrise Workbench: Help -> Install new software -> add repository, but I can not find those packages. Where can I load those packages? Thanks!

aravindsiv, 2017-11-06:
Sorry for jumping in, but these are software packages that you'd need to request from KUKA to provide to you, just like the Sunrise.Connectivity module. If you reach out to KUKA support, they should be able to help you out.

aravindsiv, 2017-11-10:
Hi @yajunxuejue ,

If you were able to contact KUKA to obtain the selected packages, I'd recommend closing this issue...

closed 2017-11-11 by yajunxuejue

madhukartlr, 2019-07-01:
> Sorry for jumping in, but these are software packages that you'd need to request from KUKA to provide to you, just like the Sunrise.Connectivity module. If you reach out to KUKA support, they should be able to help you out.

I have already installed and activated the above packages in StationSetup.cat of my project. Could you tell me what else I'm missing?