IFL-CAMP/iiwa_stack #92: Joint states not being published

aravindsiv, 2017-11-06:
I'm trying to publish the joint states of my robot so that I can visualize the robot model in RViz, get transforms, etc. To set _robot_description_ on the parameter server, I figured out that I have to run the launch file in _iiwa_description_. Then, as I understand it, to ensure that the joint states are published, I have to set the parameter _publishJointStates_ to true. However, even after doing this, my joint states are being published. 

Could anybody point out where I'm going wrong? Apologies for the barrage of issues lately, hopefully this will serve as a troubleshooting guide of sorts to future users.

Thanks in advance!

SalvoVirga, 2017-11-10:
Do you want to visualize the robot state in RViz without loading the MoveIt! module?
If no (so if also launch MoveIt! is ok), [this launch file](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_moveit/launch/moveit_planning_execution.launch) has everything you need, or you could make your own launch files starting from that one.
Else (so if you really don't want to even load MoveIt!), the robot application can already publish joint_states, there is a method ["getPublishJointStates" in iiwaConfiguration](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/iiwaConfiguration.java#L284) that is used in [ROSBaseApplication](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSBaseApplication.java#L200) to decide if joint_states should be published.

Mind that the robot_description lives under the "/iiwa" name space (i.e. /iiwa/robot_description) and does basically everything related to the robot. You can follow how that launch file that I mentioned loads internally the model and spawn the robot_state_publisher via [iiwa_control.launch](https://github.com/SalvoVirga/iiwa_stack/blob/master/iiwa_control/launch/iiwa_control.launch)

SalvoVirga, 2017-12-19:
is this fixed?

aravindsiv, 2017-12-19:
Hi @SalvoVirga ,

Thanks for your help with this, I was able to use that. Sorry for not closing this earlier.

closed 2017-12-19 by aravindsiv