IFL-CAMP/iiwa_stack #93: ROSSmartServo IK error

aravindsiv, 2017-11-10:
Hi,

When I send a PoseStamped message to /iiwa/command/CartesianPose, I get the following error on my SmartPAD:

[image]

I'm guessing this is probably because the robot cannot physically reach the queried pose. However, I'm able to manually guide the arm to that pose (with some margin of error, of course). 

Any idea what could be going wrong here?

aravindsiv, 2017-11-10:
Since these problems arose after we attached a gripper to the media flange, I'd like to add the following notes:

I have attached a gripper to the media flange (only for power). I am not using the communication port since right now, it has been easier for me to send the gripper commands from my ROS terminal directly. I have not set up the media flange on Sunrise in any way (such as adding a tool, etc.)

Thanks in advance!

SalvoVirga, 2017-11-10:
I would strongly suggest you to setup the tool in Sunrise, that we you can directly send commands in tool space, [here](https://github.com/SalvoVirga/iiwa_stack/wiki/attachtool) you should find how to do so also creating an URDF of the robot+tool.

Since you did not setup the tool so far, I imagine your are sending cartesian poses in the robot coordinate system. When you say that you can manually move the robot to the poses you are sending, I imagine you are printing out what you are sending to check that, be sure to check the exact values you are sending, print them right before the call to the robot. Did you also take into account the orientation of the pose that you are sending when checking?

aravindsiv, 2017-11-10:
Yes, I am sending cartesian poses in the robot coordinate frame.

I manually move my robot to a point (x,y,z,[x,y,z,w]) (I get this pose using /iiwa/state/CartesianPose). Now say the quaternion corresponds to a rpy of (-178,2,-178).

Now, say I want to use /iiwa/command/CartesianPose to move my robot to a point (x,y,z,[x',y',z',w']), where the quaternion [x',y',z',w'] corresponds to a rpy of (-180,0,-180), which, if I'm not wrong, should also be reachable.

I first move the robot to some initial home position, and send the command. And it fails.

aravindsiv, 2017-11-14:
Hi @SalvoVirga ,

I setup MoveIt!, and when I'm using the RRTConnectkConfigDefault planner with the TRAC-IK Kinematics Plugin (which I believe is supposed to be more accurate), this issue still persists, which amazes me. 

Moreover, I suspect that it occurs only if I ask the end effector to align itself along one of the axes of the robot. Could this have anything to do with it?

SalvoVirga, 2017-12-19:
Sorry for the radio silence, do you have any update on this?

aravindsiv, 2017-12-19:
This issue still persists, but I still don't have a concrete answer to why this is happening. I will offer an update soon, but feel free to close this issue.

921218316, 2020-10-27:
I met the same problem. Did you solve the problem? @aravindsiv @SalvoVirga