IFL-CAMP/iiwa_stack #95: Storing my own functions in a class to move iiwa

GitDavid18, 2017-12-05:
Hey!
I am trying to create my own class, where I can store functionalities I want to implement.
For Example move to a certain position/frame that I have stored on the iiwa. Therefore I edited the subscriber node to receive my messages via ROS. Sending messages and making the robot react to it in the ROSSmartServo class is no problem. But I do not want to store all my functionalities I want to implement in the ROSSmartServo class but want to store them in a different class file. Therefore I created a class that extends the ROSSmartServo class and all I want it to do is print out a message on the handheld of the iiwa but instead it is throwing a NullPointerException.

Long story short: Could you tell me how to store functions in a different class and just call them from ROSSmartServo?

Do I have to initialize the robot in my new class as well?? Or do I have to pass certain values to my newly created class? Do I even have to extend the ROSSmartServo class?

Thanks a lot for your help!!

SalvoVirga, 2017-12-19:
I would not inherit from ROSSmartServo, but just have another class that contains what you need and an instance of that inside ROSSmartServo.
For example, check how an object of type [Motions](https://github.com/IFL-CAMP/iiwa_stack/blob/development/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/Motions.java) is used inside ROSSMartServo (in the devel branch for the moment!). Of course, you will need to pass around the needed objects to that class.

GitDavid18, 2017-12-19:
Thanks for your help! I solved the problem by creating a new object of my customClass in ROSSmartServo and handed it the ROSSmartServo object: customClass name = new customClass(this)
And the constructor in customClass looks like this: public customClass(ROSSmartServo name){ROSSmartServo servo = name}
Now I can access everything thats stored in the original ROSSmartServo object by using servo.getLogger.info("...") for a message for example

closed 2017-12-19 by GitDavid18